goodrobots/maverick #220: Detect ocam/cameras other than video0

fnoop, 2016-12-11:
If MFC or other v4l devices present, camera isn't guaranteed to be video0.  Still need to detect it both in puppet maverick_baremetal::peripherals classes and in maverick-visiond.

siliconalchemy, 2016-12-14:
Default maverick-visiond.conf sets /dev/video0:
# Camera device, almost always /dev/video0
camera = /dev/video0

Perhaps this should autodetect.

fnoop, 2016-12-14:
Default maverick-visiond.conf sets /dev/video0:
```
# Camera device, almost always /dev/video0
camera = /dev/video0
```

Perhaps this should autodetect.

fnoop, 2016-12-14:
Added logic to maverick-visiond to scan through v4l devices and pick the camera device.  Changed default config to autodetect camera.
Also added logic to autodetect MFC encoder device.

commit 02cee384 (2016-12-15): Autodetect camera device for visiond, add initial hardware encoder detection, fixes #220
closed 2016-12-15 by fnoop