goodrobots/maverick #233: libtbb-dev install error on raspbian

fnoop, 2016-12-23:
```
Error: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install libtbb-dev' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
Package libtbb-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libtbb-dev' has no installation candidate
Error: /Stage[main]/Maverick_vision::Opencv/Package[libtbb-dev]/ensure: change from purged to present failed: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install libtbb-dev' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
Package libtbb-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libtbb-dev' has no installation candidate
```

fnoop, 2016-12-23:
tbb not available in raspbian.

fnoop, 2016-12-23:
http://hackaday.com/2016/12/01/running-intel-tbb-on-a-raspberry-pi/
https://www.theimpossiblecode.com/blog/intel-tbb-on-raspberry-pi/

abhiTronix, 2018-04-08:
@fnoop  https://github.com/abhiTronix/TBB_Raspberry_pi Solution :)

fnoop, 2018-04-12:
@abhiTronix How much difference does it make on a raspberry?

abhiTronix, 2018-04-12:
2x~3x performance increase when used in conjunction with OpenCV (compiled
along with TBB), also fixes performance related problems in face
recognition (dlib) .All n all boost in overall OpenCV related applications
performance in my Rpi. More info. on my git.

fnoop, 2018-04-29:
@abhiTronix Where are the binaries?

abhiTronix, 2018-04-30:
@fnoop  Latest .deb files and full support is only provided through email (abhi.una12@gmail.com) . 

> Consider supporting my countless hours of hardwork and helping me out by making small Donation for my ongoing Independent A.I. Research. Thankyou

.

fnoop, 2018-04-30:
Are you kidding me?  You're trying to sell opensource software?
Closed

closed 2018-04-30 by fnoop

abhiTronix, 2018-04-30:
@fnoop Sorry for inconvenience caused, the files that I'm providing is best optimized in terms of speed and accuracy making OpenCV highly optimized for raspberry pi. Since Building these files took me a lot of time and precision , therefore i can't give them freely in public. Hope you can understand.

However simple unoptimized precompiled latest deb files for Rpi are already available everywhere on internet along with installation instructions associated . It's a matter of interest and passion ,So goodluck🙂️.

fnoop, 2018-04-30:
No, I don't understand.  TBB is opensource software, OpenCV is opensource software, this project Maverick is opensource software.  Everyone gives this software for free, for the greater good of human knowledge.  You setting up a fake github project and trying to sell this software is a disgrace, shame on you.

abhiTronix, 2018-04-30:
I think you're misguided by the knowledge you currently have, Firstly no one asking for money for the files it's about donating my ongoing research on A.I. as i stated, Moreover if you're so into thinking that it's fake then go on create one of your own and then give away it in public but sadly you're a pathetic lazy person . It's your narrow mindedness which make you disgrace ,you can't judge someone if don't know what he/she is capable off. 

> At last Famous lines : ' if you're good at something never do it for free'

commit 948afdd5 (2016-12-23): Disbale libtbb install attempt, fixes #233