goodrobots/maverick #271: Add rtsp server to visiond

fnoop, 2017-03-01:
Test rtsp implementation for easy video stream.
See if it can be broadcast through avahi or other mechanism.

fnoop, 2017-04-06:
Some references:
https://github.com/GStreamer/gst-rtsp-server/tree/master/examples
https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gst-rtsp-server/html/GstRTSPServer.html
http://trac.gateworks.com/wiki/Yocto/gstreamer/streaming
https://github.com/Gateworks/gst-gateworks-apps/blob/master/src/gst-variable-rtsp-server.c

fnoop, 2017-04-06:
Python refs:
https://github.com/paulmilliken/SimplePythonRtspExample/blob/master/AxisRtspSimple.py
http://stackoverflow.com/questions/36164830/gstreamer-rtsp-server-not-working-sdp-contains-no-streams

fnoop, 2017-04-09:
http://stackoverflow.com/questions/34012197/data-transfer-through-rtsp-in-gstreamer

Kiwa21, 2017-05-13:
https://github.com/Gateworks/gst-gateworks-apps/blob/master/src/gst-variable-rtsp-server.c

fnoop, 2017-11-28:
https://www.metal3d.org/ticket/serveur-rstp-avec-gstreamer-et-python

fnoop, 2017-11-28:
https://github.com/mpromonet/v4l2rtspserver

fnoop, 2017-11-28:
http://gstreamer-devel.966125.n4.nabble.com/How-to-create-one-pipeline-for-both-audio-video-stream-and-only-video-stream-td4681575.html

fnoop, 2017-11-29:
First run at rtsp server.  RTSP Server is working and supports multiple clients, but on the last client disconnecting visiond segfaults.  Suspect this is because it tries to tear down the pipeline, which it shouldn't do as it didn't create it.

fnoop, 2017-11-29:
Segfaulting on all client shutdown isn't ideal, but the systemd service that controls visiond relaunches on failure, so practically speaking the user shouldn't notice - the rtsp service is immediately available again.

closed 2017-11-29 by fnoop