goodrobots/maverick #304: How to connect joule dev kit to pixhawk

fnoop, 2017-03-09:
Work out gpio pins, levels, linux devices etc

fnoop, 2017-03-09:
Conflicting info on the gpio levels:
https://communities.intel.com/thread/111850
https://communities.intel.com/thread/111266
https://communities.intel.com/thread/109713

fnoop, 2017-03-09:
GPIO pinouts:
http://www.intel.com/content/www/us/en/support/boards-and-kits/intel-joule-kits/000022494.html

fnoop, 2017-03-09:
Hmm
https://communities.intel.com/message/436334#436334

fnoop, 2017-03-09:
Aha!
https://github.com/intel-iot-devkit/mraa/issues/612

fnoop, 2017-03-10:
So yes UARTs on the expansion board work fine.  Connected pixracer to pins 22/24 (tx/rx from perspective of joule, rx/tx from perspective of pixracer) and ground to pin 36, works perfectly.
Will also try ttys2 which is mini-usb ftdi, might be neater physical integration.  Note ttys0 also has uart flow control pins exposed which might be nice if higher speed is required.

closed 2017-03-10 by fnoop

lamping7, 2017-03-27:
I followed the links you posted and cannot get this to work.
`x = mmra.Uart("/dev/ttyS1")` fails with an initialization error

Ultimately I'm looking to interface with the pixhawk through mavros. Can you share how you got this to work?

fnoop, 2017-03-28:
@lamping7 Hi, the easiest way to test it is probably with mavproxy, which autodetects uart.  It works fine for me with the pixhawk and pixracer using pin 36 for ground (pin 6 on the telemetry port) and pins 24 and 22  connected to pin 2/3 on the telemetry port (can't remember which way round, try both).  Make sure to set the baud correctly in the pixhawk firmware for serial1 (or serial2/4 depending on which telemetry port you're using) - 115200 is a good start.  Then start mavproxy something like: `mavproxy.py --master /dev/ttyS1 --baud 115200`

lamping7, 2017-03-28:
@fnoop Thanks for the reply. Here is the output from mavproxy:
```
$ mavproxy.py --master /dev/ttyS1 --baud 57600
Connect /dev/ttyS1 source_system=255
Failed to connect to /dev/ttyS1 : Could not configure port: (5, 'Input/output error')
```
Looking at `/var/log/syslog` after running the above python code from my previous comment:
```
libmraa[18858]: libmraa version v1.5.1 initialised by user 'uas' with EUID 1000
libmraa[18858]: joule: Failed to detect i2c buses, making wild assumptions!
libmraa[18858]: imraa: Failed to open lock file
libmraa[18858]: libmraa initialised for platform 'Intel GT Tuchuck' of type 13
libmraa[18858]: uart: tcgetattr(/dev/ttyS1) failed: Input/output error
```
Did you have to set anything in the BIOS to configure UART? Could you provide me with more info or your system? What is your BIOS version and linux kernel version (`uname -a`)

More info about mine:
kernel version: 4.4.0-67-generic 88-Ubuntu SMP Wed Mar 8 16:34:45 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

fnoop, 2017-03-28:
Interesting.  I don't think I changed anything in the BIOS (but might have).
```
[dev] [mav@fnoop-joule ~/data/config/mavlink]$ sudo dmidecode |grep Version
	Version: GTPP174A.X64.0143.B30.1612190410
[dev] [mav@fnoop-joule ~/data/config/mavlink]$ uname -a
Linux fnoop-joule 4.4.0-1000-joule #0+joule12-Ubuntu SMP PREEMPT Fri Jan 6 04:28:41 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
Looks like you're using the standard ubuntu rather than the joule spin?
https://developer.ubuntu.com/core/get-started/intel-joule

lamping7, 2017-03-28:
I've got Ubuntu 16.04 Server on it, which is why it has the generic kernel. I didn't want the desktop environment. I'll up the BIOS from 161 to 174 as seen on your system and recommended by Canonical and go from there. Thank you for the help!

fnoop, 2017-03-28:
Ah OK does everything else seem to work?  My next task is trying to get opengl/opencl working, which seems like a much bigger challenge.

lamping7, 2017-03-28:
OpenCL for use with OpenCV? Good luck. The Joule is terrible in terms of documentation and has a very small community behind it. Your project looks very interesting though.

My initial look into its GPIO resulted in only getting it work with the mraa library using sudo, which is unsatisfactory and will require more research. I've also seen a good amount of thermal CPU warnings in the terminal with the Joule stationary on the bench with little load (with heatsink, no fan). In my opinion, it is a board with an OK processor, but not much else.

Also, upgrading the BIOS didn't do the trick. I'll look into the kernel. Thanks again!

fnoop, 2017-03-29:
The joule has a pretty capable gpu apparently - I saw a figure of >200gflops somewhere, so it's worth trying to get it to work!  The joule has some superb documentation but it's really all down at the hardware/integration level rather than up at the software level, which i'll agree is pretty poor.  Particularly ubuntu support, don't understand why they persevere with yocto if they actually want people to use it..
I'm about to try ubuntu 16.10 desktop+server, perhaps the newer kernel will support the UARTs - Intel do have a good track record of getting hardware support into upstream kernel.

lamping7, 2017-03-29:
Just to close the loop in case anyone finds this: You need the special Joule kernel, which is not public yet.-> http://askubuntu.com/questions/878880/where-is-the-source-for-the-4-4-0-1000-joule-kernel

fnoop, 2017-03-30:
@lamping7 Interesting, thanks.  Have queried why the source isn't available, as that's obviously a GPL requirement..

lamping7, 2017-04-01:
@fnoop Something weird is going on with that. That question I linked to above had comments removed, then they deleted it altogether. See my comments (that include past screenshots) here: https://communities.intel.com/thread/111669?start=15&tstart=0

moonsummer, 2018-01-10:
how can run maverick with raspberry？

fnoop, 2018-01-10:
@moonsummer How is your question related to the Joule?