goodrobots/maverick #388: Lots of odroid gstreamer warnings with visiond

fnoop, 2017-05-12:
```
[dev] [mav@maverick-odroidxu4 ~]$ maverick-visiond start

** (gst-plugin-scanner:6113): CRITICAL **: gi.repository.Gst is no dict
sys:1: Warning: cannot register existing type 'GstSoupHTTPSrc'
sys:1: Warning: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed
sys:1: Warning: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sys:1: Warning: cannot register existing type 'GstSoupHttpClientSink'

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sys:1: Warning: cannot register existing type 'GstViewfinderBin'

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sys:1: Warning: cannot register existing type 'GstOss4Sink'

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sys:1: Warning: cannot register existing type 'GstPulseSink'

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sys:1: Warning: cannot register existing type 'GstEncodeBin'

(gst-plugin-scanner:6113): GStreamer-CRITICAL **: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused

(gst-plugin-scanner:6113): Clutter-CRITICAL **: Unable to initialize Clutter: Could not initialize Gdk

(gst-plugin-scanner:6113): Clutter-Gst-CRITICAL **: Unable to initialize Clutter
```

fnoop, 2017-05-12:
This displays when first running gstreamer or after removing ~/.cache/gstreamer-1.0.  Might indicate the wrong (system) gstreamer libraries are being picked up.

fnoop, 2017-05-12:
GST_PLUGIN_PATH is set OK:
[dev] [mav@maverick-odroidxu4 ~]$ env |grep GST
GST_PLUGIN_PATH=/srv/maverick/software/gstreamer/lib/gstreamer-1.0

fnoop, 2017-05-12:
So yes these warnings are because after scanning GST_PLUGIN_PATH, gstreamer then scans GST_PLUGIN_SYSTEM_PATH which picks up any system installed gstreamer plugins.  This is undesirable - maverick gstreamer should be constrained to it's own plugins.  If we set GST_PLUGIN_SYSTEM_PATH to GST_PLUGIN_PATH it should prevent this.

fnoop, 2017-05-12:
The remaining error:
```
** (gst-plugin-scanner:452): CRITICAL **: gi.repository.Gst is no dict
```
only shows once while building the cache and doesn't seem to have a lasting effect.

closed 2017-05-12 by fnoop

sammck, 2018-08-07:
@fnoop If your experience is mine, the reason the error only displays once is because the plugin that caused the error is blacklisted, and is not ever checked again until you delete the registry cache. try 

> get-inspect-1.0 -b

to confirm...

fnoop, 2018-08-09:
@sammck thanks for info.  I gave up on odroid a while back, have not had this problem with any other platform so far!

commit 03fbf8bb (2017-05-12): Fix gstreamer plugin paths, don't pick up system gstreamer plugins, fixes #388