goodrobots/maverick #389: visiond not accepting pipeline_override

fnoop, 2017-05-12:
maverick-visiond doesn't seem to accept pipeline_override to override the auto pipeline.

fnoop, 2017-05-12:
@patrickpoirier51 - I've uncommented pipeline_override config entry in ~/data/config/vision/maverick-visiond.conf and it definitely picks it up:
```
[dev] [mav@maverick-odroidxu4 ~/var/log/vision]$ more maverick-visiond.log
2017-05-12 18:11:06,169 - INFO - Starting maverick-visiond
2017-05-12 18:11:06,171 - INFO - Config read from /srv/maverick/data/config/vision/maverick-visiond.conf
2017-05-12 18:11:06,171 - INFO - Manual Pipeline Construction
2017-05-12 18:11:06,172 - INFO - Creating pipeline from config: v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! autovideo
convert ! vaapih264enc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.111 port=5000 sync=false
```
Could you please try again?

patrickpoirier51, 2017-05-12:
Yes it works fine

fnoop, 2017-05-13:
Thanks for testing :)

closed 2017-05-13 by fnoop