goodrobots/maverick #396: SITL Parameters and Options

SamPes, 2017-05-13:
With SITL, the 5 most used parameters that should be easy to configure (in the webgui eventually) are: 
1) set aircraft type that also selects the appropriate firmware: quadplane, plane using arduplane firmware, and copter etc using arucopter firmware
2) by location: select between CMAC and other locations from a list (list is in locations.txt) along with the ability to point on a map and/or manually add to that locations list easily. Alternatively using the current home position on the FC. 
3) Set: Wind direction in degrees heading
4) Set: Wind speed in m/s
5) Wipe virtual EEPROM of parameters using: sim_vehicle.py -w

Others options like linking to external simulators can be added later, but running mavcesium locally would likely be good enough to visualise the aircraft flight and mission using a web interface.
Ideally all these items would be given their own tab/config/widget in the webgui.

fnoop, 2017-05-14:
1. This must currently be set as a maverick parameter, because changing it involves recompiling firmware, adding jsbsim.  The parameter is "maverick_dev::ardupilot::ardupilot_vehicle" and should be 'copter', 'plane', or 'rover'.

fnoop, 2017-05-14:
SITL is started using sim_vehicle.py, but starts mavproxy separately (if at all).  Need to add a config file that influences the sim_vehicle.py startup parameters.

fnoop, 2017-05-15:
2. and 5. and more satisfied by ~/data/config/mavlink/sitl.conf.
Wind is set by a parameter within SITL.  In-SITL options are a different matter and should probably be dealt with by GCS or a web equivalent, which is out of scope of this issue.

closed 2017-05-15 by fnoop