goodrobots/maverick #424: Change ROS source install to desktop

fnoop, 2017-05-17:
ROS source install is currently ros_comm, which does not include rviz and other desktop tools.

khancyr, 2017-06-27:
@fnoop the recommended default for ROS should be ros-base . That way you have all stuff to start but without graphical thing that won't work on small platform

fnoop, 2017-06-27:
Yes you're right I think!  All of the downloadable Maverick OS images are designed to be desktop capable/useful, so I thought adding a ROS desktop install would be useful for rviz etc.  However it drags in a whole crapload of dependencies including gazebo.  Which would be fine except they take up a huge amount of space and push the OS images over an 8Gb boundary, which seems like a sensible maximum.

closed 2018-01-22 by fnoop