goodrobots/maverick #437: WifiBroadcast

terial, 2017-05-23:
Hey fnoop, i was messaging you as nikker on the ardupilot forum. I have never used github too much so its all very new. If I am not following the process correct please let me know. 

Regarding wifibroadcast bortek seems to have taken it over and has a fork here
https://github.com/bortek/EZ-WifiBroadcast

in issue number 11 it seems like he is going to update the code to version 1.6 and also seems to be updating the wiki. I noticed there are a few more wifi dongles listed. One of them is the ubiquiti wifistation usb, i just purchased two of them. Anyways I am hoping he updates the code soon and it can be incorporated into maverick.

fnoop, 2017-05-23:
Hi @terial, thanks for the suggestion :)  I have followed the EZ-WifiBroadcast for a while but haven't seen anything sufficiently different/improved to use it (I might be missing the point!).  EZ-WifiBroadcast is really about integration, it's a preformed distribution image specifically for raspberry pi, wifibroadcast and some adapters.  It apparently contains altered wifibroadcast code but difficult to see what it is, as there isn't an updated git source for it.
As for supporting other devices, that's largely down to the kernel.  You might be lucky and find your ubiquiti usb is already supported.  I would much rather not start hacking kernels in Maverick as it leads to potential instability, but if it's straight forward and a good reason then will certainly have a look at it :)

terial, 2017-05-24:
Now that you mention it, i see the difference between wifibroadcast and EZ-wifibroadcast. I am going to try and get more involved with the project and see what changes have been made. I am hoping the ubiquiti hardware will be a good choice to test everything out. If I see the updated source I will let you know.

terial, 2017-06-17:
Hey fnoop! I have no been messing with Maverick lately but i see you have been hard at work. I have been busy with EZ-wifibroadcast... trying to learn more about it and getting my new hardware setup with it. The code on github has been updated to the latest 1.5 version. I read in the release notes and seen some things when it was installed that indicated you had some form of wifibroadcast implemented but not sure if it was from befitivi's code or bortek's EZ-wifibroadcast. Anyways just wanted to let you know. https://github.com/bortek/EZ-WifiBroadcast

fnoop, 2017-06-18:
Hey @terial :)  It's difficult to see from their github what sort of modifications/improvements they've made to wifibroadcast itself as the commits are really poorly documented/unclear (don't look at my commits ;) . 
I'll have a closer look and see if it's worth moving from the vanilla sources, I still think that pretty much all the work in EZ-WifiBroadcast is in the integration/building of a raspberry image, including kernel mods.  Which is great, but they don't really have any build process to follow so really difficult to use the work in other projects.  I have little appetite to start modding the kernels as that is more likely to lead to instabilities than it is to help people, overall.
But what they're doing is really cool and interesting, so I'll take a further look when I get a few minutes :)

terial, 2017-06-18:
I agree with you. EZ WiFi broadcast seems to be very tightly integrated to the point of almost being difficult to modify for other configurations. I ended up getting Alfa wifi cards  and have setup Maverick with wlan0 as an access point on 2.4 and the Alfa card on bcast1 on 5ghz. Hopefully will be testing it all out in a few days. Should this  issue be closed now? Thanks for your time and effort!

fnoop, 2017-06-18:
That's great!  Please let us know how you get on :) I've used alfa cards in the past as well because they don't need any kernel mods and are very strong out of the box.  Might as well keep this open, I will take a further look when I get a bit of time as this is a really cool tech.

terial, 2017-06-18:
That's great to hear, I was hopping I would not have to mess with anything related to the kernel. Will keep you posted.

cglusky, 2018-03-29:
Just getting to try the wifibroadcast code.  Seems to be getting harder to find the off the shelf USB dongles listed in the wiki.  Unless you want to run 2.4GHz WiFi.  That is likely a bad idea if/until RC performance is tested.  Long thread going on RC Groups.  Did not read the entire thing.  Assuming those running 2.4 WiFi are also using a joystick vs 2.4 RC gear.  Not sure.

I have the wifi equipment to test.  On The List for Dev Frame demo build.

Also, a new extension for wifibroadcast  https://github.com/seeul8er/DroneBridge

According to issue queue should be more work and a new  DroneBridge  release next week.

fnoop, 2018-03-30:
In practice 2.4ghz wifi doesn't seem to actually have much effect on RC, if RC is spread spectrum.  Both sides detect interference and use other channels if required.

cglusky, 2018-03-31:
Depends on the equipment.  Older Spektrum DSM2 not a good idea at all.  DSMX is better.  Newer FHSS stuff may actually do OK, but cause issues for WiFi.  Either way you are running a signal/noise risk running 2.4 WiFi with 2.4 RC that I prefer to avoid without much testing.  Makes me nervous.  

There is now a different Wifibroadcast project.  It also happens to be the one linked to from the PX4 wiki so we have potential for confusion:  https://dev.px4.io/en/qgc/video_streaming_wifi_broadcast.html

fnoop, 2018-04-01:
Oh, that's fantastic!  I'll add support for this new code, hopefully we'll get lower latency from it, and the old code was abandoned years ago anyway.
WRT inteference, I've been running high power wifi alongside X8R/X4R receivers for years and never had a problem.

closed 2018-04-29 by fnoop