goodrobots/maverick #445: I was not able to create a AP Interface

mengelh, 2017-05-28:
"maverick_network::interfaces": {
        	"wlan0": {
			    "type": "wireless",
			    "mode": "ap",
			    "macaddress": "b8:27:eb:a8:1d:bb", // Mac of my PI 3 Wifi Adapter
			    "ssid": "Piglet",
			    "driver": "nl80211",
			    "channel": 1,
			    "hw_mode": "g",
			    "disable_broadcast_ssid": false,
			    "dhcp_range": "192.168.10.10,192.168.10.50",
			    "dhcp_leasetime": "24h",
			}
	},

Error: Invalid parameter driver on Maverick_network::Process_interface[wlan0] on node maverick-raspberry.fritz.box
Wrapped exception:
Invalid parameter **driver**
Error: Invalid parameter driver on Maverick_network::Process_interface[wlan0] on node maverick-raspberry.fritz.box

My questions:
- What am I doing wrong
- How do I set the password
- How to combine a AP Interface withe a other Wifi, by using a second stick)

fnoop, 2017-05-29:
@mengelh Hi, thanks for reporting this, silly mistake on my part.  It should be fixed now, and I've updated the docs here:  
http://fnoop.github.io/maverick/#/modules/network

Theoretically, you should be able to now do:
```
maverick self-update
maverick configure
```
And hey presto it should work :)  Note you might want to set your psk and apaddress as per the docs first.

mengelh, 2017-05-29:
thanks a lot, I will test it out after work :)

mengelh, 2017-05-29:
AP Still do not com up.

Localconf.json

// Sample/default config file for Raspberry Pi platform
// Activated by changing hostname to 'maverick-raspberry' and running 'maverick configure'
{

	// Set defaults for wifi.  If set, any wifi interface detected will be automatically activated with these settings.
	// These setting can be overridden in the per-interface settings.
	// "wifi_ssid": 		"MySSID",
	// "wifi_passtype": 	"wpa2",
	// "wifi_passphrase":	false, // unencrypted wifi passphrase, strongly recommended to use psk below instead
	// "wifi_psk": 		"averylonggeneratedkey", // must be encrypted psk created using 'wpa_passphrase MySSID MyPassphrase'

	// Note a fallback generic eth0 and wlan0 config is always created in addition to these settings
	"maverick_network::interfaces": {
		    "wlan1": {
        	    "type":	"wireless",
        	    "macaddress": "24:05:0f:6b:7a:e1", // Stick
        	    "ssid":	"wlan_matti",
        	    "psk":	"1234" // must be encrypted psk created using 'wpa_passphrase MySSID MyPassphrase'
        	},
        	"wlan0": {
			    "type": "wireless",
			    "mode": "ap",
			    "macaddress": "b8:27:eb:a8:1d:bb", // PI 3 Wifi adapter
			    "ssid": "Piglet",
			    "psk": "1234",
			    "driver": "nl80211",
			    "channel": 1,
			    "hw_mode": "g",
			    "disable_broadcast_ssid": false,
			    "apaddress": "192.168.10.1",
			    "dhcp_range": "192.168.10.10,192.168.10.20",
    			"dhcp_leasetime": "24h"
			}
	},

	"maverick_desktop::enable": false, // Desktop depends on type of raspbian used

	// Flight Controller (fc) mavlink setup
	// "maverick_fc::mavlink_proxy": "mavproxy", // Default mavlink proxy type to use
	// "maverick_fc::mavlink_active": true, // Whether the mavlink proxy is active or not
	// "maverick_fc::mavlink_input": "/dev/ttyAMA0", // Input source for mavlink proxy, how the Companion Computer connects to the Flight Controller
	// "maverick_fc::mavlink_baud": "115200", // Input baud rate for UART connections
	// "maverick_fc::rosmaster_active": true, // Whether ROS master for fc is active or not
	// "maverick_fc::mavros_active": true, // Whether Mavros for fc is active or not - requires fc rosmaster to be active
	
	"maverick_dev::sitl::mavros_startup_delay": 20,
	"maverick_fc::mavros_startup_delay": 20,

	// Dev SITL mavlink setup - note this requires dev environment to be active
	// "maverick_dev::sitl::mavlink_proxy": "mavproxy", // Default mavlink proxy type to use
	// "maverick_dev::sitl::sitl_active": true, // Whether SITL is active or not
	// "maverick_dev::sitl::mavlink_active": true, // Whether mavlink proxy is active or not
	// "maverick_dev::sitl::rosmaster_active": true, // Whether ROS master for sitl is active or not
	// "maverick_dev::sitl::mavros_active": true, // Whether Mavros for sitl is active or not - requires sitl rosmaster to be active
	
	// "maverick_vision::visiond::active": true,
	// "maverick_vision::vision_landing::active": false,
	
	// Add classes here
	"classes":		[
	    "maverick_desktop",
	    "maverick_network"
	]
	
}

fnoop, 2017-05-29:
So does the configure run go OK?  Have you rebooted?  What is in /etc/hostapd/hostapd.conf?

mengelh, 2017-05-29:
/etc/hostapd/hostapd.conf
# Maverick hostapd configuration
# This file is managed by maverick/puppet, do not alter by hand

interface=wlan0
driver=nl80211
ssid=Piglet
channel=1
hw_mode=g
ignore_broadcast_ssid=false

#Sets authentication algorithm
#1 - only open system authentication
#2 - both open system authentication and shared key authentication
auth_algs=1

#####Sets WPA and WPA2 authentication#####
#wpa option sets which wpa implementation to use
#1 - wpa only
#2 - wpa2 only
#3 - both
wpa=2
#sets wpa passphrase required by the clients to authenticate themselves on the network
wpa_psk=1234
#sets wpa key management
wpa_key_mgmt=WPA-PSK
#sets encryption used by WPA
wpa_pairwise=TKIP CCMP
#sets encryption used by WPA2
rsn_pairwise=CCMP

 /etc/network/interfaces 
# Interface lo
auto lo
iface lo inet loopback
# Interface wlan0
auto wlan0
allow-hotplug wlan0
iface wlan0 inet static
    address 192.168.10.1
    netmask 255.255.255.0
    wireless-mode master
# Interface eth0
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
# Interface wlan1
auto wlan1
allow-hotplug wlan1
iface wlan1 inet dhcp
    wpa-ssid wlan_matti
    wpa-psk  1234

mengelh, 2017-05-29:
No issues while the configurations and rebooted directly after configuration. 

You can see everything seems to look fine.

fnoop, 2017-05-29:
Thanks for the info.  I don't have a Pi 3 but I do have a Zero W which I believe has the same chipset, I'll test it this evening.

mengelh, 2017-05-29:
I have made some tests, if I use:

	"maverick_network::interfaces": {
		    "wlan1": {
        	    "type":	"wireless",
        	    "macaddress": "24:05:0f:6b:7a:e1", // Stick
        	    "ssid":	"wlan_matti",
        	    "psk":	"9075059202412158#*" // must be encrypted psk created using 'wpa_passphrase MySSID MyPassphrase'
        	},
       "wlan0": {
	            "type": "wireless",
	            "mode": "ap",
	            //"ssid": "Piglet",
	            //"psk": "4dm7QIgn",
	           "macaddress": "b8:27:eb:a8:1d:bb"
			}
	},

It works. But if I add any custom Parameter eg. ssid or psk, it dos not work ....

fnoop, 2017-05-29:
Just to confirm, for the psk you're putting the very long hex string output by `wpa_passphrase <ssid> <passphrase>`?

mengelh, 2017-05-29:
Excuse me, I did not see the forest in front of the trees ... now it works

closed 2017-05-29 by mengelh

fnoop, 2017-05-29:
Great, thanks for the testing and feedback :)