goodrobots/maverick #466: Bootstrap Error on clean Trusty 14.04

Kiwa21, 2017-06-09:
Hardware : http://www.gateworks.com/product/item/ventana-gw5510-single-board-computer
OS : Ubuntu 14.04 LTS (Trusty Tahr)  + gstreamer-imx for IPU/VPU/GPU hw video acceleration 
Kernel : Gateworks 3.14 kernel with modules 

After installing a Ubuntu 14.04 on a SD card for a Gateworks GW5510 with the method described here (rootfs) :
`
http://trac.gateworks.com/wiki/linux/blockdev
`

After cloning the git rep and trying to install Maverick with the bootstrap method : 
./bin/maverick --env=bootstrap configure

I have the following error :  
`
Maverick - UAV Companion Computer System

Maverick Environment: bootstrap
Proceeding to update system configuration - please be patient, this can take a while..

[ 737.355768] Unhandled fault: external abort on non-linefetch (0x018) at 0x76eb9000
[ 745.359956] Unhandled fault: external abort on non-linefetch (0x018) at 0x76e15000
Warning: Could not retrieve fact fqdn
Error: Could not find class base for ventana-trusty on node ventana-trusty
Error: Could not find class base for ventana-trusty on node ventana-trusty 
 `

fnoop, 2017-06-09:
Initial error also:
```
Error: Could not parse for environment bootstrap: Is a directory - /maverick/manifests on node ventana-trusty
Error: Could not parse for environment bootstrap: Is a directory - /maverick/manifests on node ventana-trusty
```
Need to specify manifests/init.pp when calling puppet, not just the directory - that wasn't supported until puppet 3.5.0.

fnoop, 2017-06-10:
Problems now with not picking up hashes from python facts, probably puppet isn't picking up puppet.conf when not run from /srv/maverick.  Probably needs to be called differently for <3.5.0.

fnoop, 2017-06-10:
Problem with structured fact in facter 2.4:
```
Could not retrieve fact='python_modules', resolution='<anonymous>': undefined method `lines' for nil:NilClass
```

Kiwa21, 2017-06-10:
Should I create the /srv directory on the clean OS then run git clone from there so I get maverick into /srv/maverick before running ./bin/maverick --env=bootstrap configure ?

fnoop, 2017-06-12:
https://tickets.puppetlabs.com/browse/PUP-7523

fnoop, 2017-06-12:
Refactoring entire codebase to puppet 4, will address this issue.
#467

fnoop, 2017-06-12:
Input group doesn't exist:
```
or: Could not create user mav: Execution of '/usr/sbin/useradd -c Maverick user -g 6789 -G dialout,input,video -d /srv/maverick -p $6$YuXyoBZR$cR/cNLGZV.Y/nfW6rvK//fjnr84kckI1HM0fhPnJ3MVVlsl7UxaK8vSw.bM4vTlkF4RTbOSAdi36c5d2hJ9Gj1 -s /bin/bash -u 6789 mav' returned 6: useradd: group 'input' does not exist
Error: /Stage[bootstrap]/Base::Users/User[mav]/ensure: change from absent to present failed: Could not create user mav: Execution of '/usr/sbin/useradd -c Maverick user -g 6789 -G dialout,input,video -d /srv/maverick -p $6$YuXyoBZR$cR/cNLGZV.Y/nfW6rvK//fjnr84kckI1HM0fhPnJ3MVVlsl7UxaK8vSw.bM4vTlkF4RTbOSAdi36c5d2hJ9Gj1 -s /bin/bash -u 6789 mav' returned 6: useradd: group 'input' does not exist
Notice: /Stage[bootstrap]/Base::Hostip/Host[ubuntu.l
```

fnoop, 2017-06-12:
gid 106 already exists:
```messagebus:x:106:```

OK instead only add input group if it exists

fnoop, 2017-06-12:
Need to add systemd to ubuntu 14.04, too many thing rely on it to start backporting to upstart

fnoop, 2017-06-14:
Ensure systemd installed and authoritative after ubuntu installation, before running maverick bootstrap:
sudo apt-get -y install systemd systemd-services
Use systemd as the init, for standard ubuntu 14.04, change /etc/default/grub from:
 `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"`
to
 `GRUB_CMDLINE_LINUX_DEFAULT="init=/lib/systemd/systemd"`
Update grub:
```
sudo update-grub
sudo reboot
```

fnoop, 2017-06-14:
14.04 bootstrap basically works now, a few issues left to deal with
```
Error: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install python-wxgtk3.0' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-wxgtk3.0
E: Couldn't find any package by regex 'python-wxgtk3.0'
Error: /Stage[main]/Maverick_mavlink/Package[python-wxgtk3.0]/ensure: change from purged to present failed: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install python-wxgtk3.0' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python-wxgtk3.0
E: Couldn't find any package by regex 'python-wxgtk3.0'
```

```
Error: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install libtool-bin' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
Package libtool-bin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtool:i386 libtool

E: Package 'libtool-bin' has no installation candidate
Error: /Stage[main]/Maverick_vision::Gstreamer/Package[libtool-bin]/ensure: change from purged to present failed: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install libtool-bin' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
Package libtool-bin is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtool:i386 libtool

E: Package 'libtool-bin' has no installation candidate
```

```
Error: '/bin/systemctl set-default multi-user.target' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_desktop/Exec[disable-desktop-target]/returns: change from notrun to 0 failed: '/bin/systemctl set-default multi-user.target' returned 1 instead of one of [0]
```

```
Notice: /Stage[main]/Maverick_ros/Exec[catkin_rosinstall]/returns: [std_msgs] Fetching https://github.com/ros-gbp/std_msgs-release/archive/release/kinetic/std_msgs/0.5.11-0.tar.gz (version std_msgs-release-release-kinetic-std_msgs-0.5.11-0) to /srv/maverick/var/build/ros_catkin_ws/src/std_msgs
Error: '/usr/bin/rosinstall_generator ros_comm --rosdistro kinetic --deps --wet-only --tar > kinetic-ros_comm-wet.rosinstall && /usr/bin/wstool init -j1 src kinetic-ros_comm-wet.rosinstall' returned 1 instead of one of [0]
```
ROS needs refactoring for 14.x

fnoop, 2017-11-12:
14.04 didn't use systemd - used the old upstart/init systems.  Maverick is really focused on systemd support.  Support for 14.04 generally has been added now as of version 1.1.4 but services won't be controlled.

closed 2017-11-12 by fnoop