goodrobots/maverick #483: Add support for PX4

fnoop, 2017-06-16:


cglusky, 2018-01-09:
some nice detailed info on PX4 offboard...

https://uav-lab.org/tutorials/software/

fnoop, 2018-01-11:
https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html
https://dev.px4.io/en/simulation/

fnoop, 2018-01-11:
Basic support for compiling px4 added. Off by default, localconf param to enable:
```"maverick_dev::px4": true```
Compiles sitl firmware by default.  To enable cross compile for pixhawk etc, set localconf param:
```	"maverick_dev::px4::cross_compile": true```

fnoop, 2018-01-12:
Looks like px4 doesn't have a built in SITL simulator like ardupilot, which will make running out of the box SITL difficult.

fnoop, 2018-01-12:
http://discuss.px4.io/t/lightweight-or-command-line-sitl-simulation-options/2285/8

SamuelDudley, 2018-02-18:
Running gazebo headless:
http://answers.gazebosim.org/question/18553/option-to-disabe-rendering-in-gazebo-8/

closed 2018-03-01 by fnoop