goodrobots/maverick #486: Add Documentation: How to use Raspberry Pi Camera with Gstreamer

mengelh, 2017-06-18:
I want to use my Pi Camera as video source for FPV using Gstreamer. But I don't understand how to configure maverick to do so.

fnoop, 2017-06-18:
Hi @mengelh :)  Are you using visiond?  It should pick up the raspberry camera automatically on /dev/video0, but you have to tell it what to do with the output (output setting in /srv/maverick/data/config/vision/maverick-visiond.conf).
Does the log show up that it's correctly detected the camera? (/srv/maverick/var/log/vision/maverick-visiond.log)

mengelh, 2017-06-24:
The log seems shows that the camera is detected and the stream is started. But if I Start QGroundcontrol (UDP 5600) I could not reserve the stream.
-----------------------------------------------------------------------------------------------------------
/srv/maverick/var/log/vision/maverick-visiond.log:
---------------------------------------------------
2017-06-24 16:22:34,502 - INFO - Starting maverick-visiond
2017-06-24 16:22:34,508 - INFO - Config read from /srv/maverick/data/config/vision/maverick-visiond.conf
2017-06-24 16:22:34,513 - DEBUG - V4l2 device input: Camera 0:2
2017-06-24 16:22:34,514 - INFO - v4l2 device /dev/video0 is a camera, autoselecting
2017-06-24 16:22:34,515 - DEBUG - V4l2 device input: Camera 0:2
2017-06-24 16:22:34,515 - DEBUG - driver: bm2835 mmal
2017-06-24 16:22:34,516 - DEBUG - card: mmal service 16.1
2017-06-24 16:22:34,516 - DEBUG - Camera control: Brightness
2017-06-24 16:22:34,517 - DEBUG - Camera control: Contrast
2017-06-24 16:22:34,518 - DEBUG - Camera control: Saturation
2017-06-24 16:22:34,519 - DEBUG - Camera control: Red Balance
2017-06-24 16:22:34,520 - DEBUG - Camera control: Blue Balance
2017-06-24 16:22:34,521 - DEBUG - Camera control: Horizontal Flip
2017-06-24 16:22:34,521 - DEBUG - Camera control: Vertical Flip
2017-06-24 16:22:34,522 - DEBUG - Camera control: Power Line Frequency
2017-06-24 16:22:34,523 - DEBUG - Camera control: Sharpness
2017-06-24 16:22:34,524 - DEBUG - Camera control: Color Effects
2017-06-24 16:22:34,525 - DEBUG - Camera control: Rotate
2017-06-24 16:22:34,525 - DEBUG - Camera format: Planar YUV 4:2:0
2017-06-24 16:22:34,526 - DEBUG - Camera format: YUYV 4:2:2
2017-06-24 16:22:34,527 - DEBUG - Camera format: 24-bit RGB 8-8-8
2017-06-24 16:22:34,527 - DEBUG - Camera format: JFIF JPEG
2017-06-24 16:22:34,528 - DEBUG - Camera format: H.264
2017-06-24 16:22:34,528 - DEBUG - Camera format: Motion-JPEG
2017-06-24 16:22:34,529 - DEBUG - Camera format: YVYU 4:2:2
2017-06-24 16:22:34,529 - DEBUG - Camera format: VYUY 4:2:2
2017-06-24 16:22:34,530 - DEBUG - Camera format: UYVY 4:2:2
2017-06-24 16:22:34,530 - DEBUG - Camera format: Y/CbCr 4:2:0
2017-06-24 16:22:34,531 - DEBUG - Camera format: 24-bit BGR 8-8-8
2017-06-24 16:22:34,531 - DEBUG - Camera format: Planar YVU 4:2:0
2017-06-24 16:22:34,532 - DEBUG - Camera format: Y/CrCb 4:2:0
2017-06-24 16:22:34,532 - DEBUG - Camera format: 32-bit BGRA/X 8-8-8-8
2017-06-24 16:22:34,535 - INFO - Camera YUV stream available, using yuv stream
2017-06-24 16:22:34,536 - INFO - Creating stream object - camera:/dev/video0, stream:yuv, pixelformat:YUY2, encoder:h264, size:(640 x 480 / 30), output:udp
2017-06-24 16:22:34,537 - INFO - Attaching input 'v4l2': /dev/video0
2017-06-24 16:22:34,565 - INFO - Attaching stream 'yuv'
2017-06-24 16:22:34,580 - INFO - Attaching encoding 'h264'
2017-06-24 16:22:34,582 - INFO - Raspberry hardware encoder detected, using omxh264enc as h264 encoder
2017-06-24 16:22:35,039 - INFO - Attaching payload 'h264'
2017-06-24 16:22:35,047 - DEBUG - h264parse element created
2017-06-24 16:22:35,058 - DEBUG - Attaching h264pay to h264parse
2017-06-24 16:22:35,060 - INFO - Attaching output 'udp', sending to 192.168.10.10:5600
2017-06-24 16:22:35,077 - INFO - Pipeline: "v4l2-source /dev/video0 ! capsfilter 'video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction)30/1' ! queue ! autovideoconvert ! raspberry-h264-encode ! h264parse ! pay0 ! udpsink"
2017-06-24 16:22:35,101 - INFO - Starting camera stream
-----------------------------------------------------------------------------------------------------------
/srv/maverick/data/config/vision/maverick-visiond.conf:
--------------------------------------------------------
[visiond]

# On/off switch for maverick-visiond, set to False to disable
enable = True

# Camera device, almost always /dev/video0
# camera_device = /dev/video0

# Camera format, if not specified here will autodetect yuv->mjpeg->h264
# format = yuv

# Pixel Format, could be (fourcc)YUV2, I420 etc
# pixelformat = I420

# Encoder, if not specified will default to h264.  Values are h264, mjpeg or none
# encoder = h264

# Resolution width of video stream, must be valid for camera
width = 640

# Resolution height of video stream, must be valid for camera
height = 480

# Framerate of video stream, must be valid for camera
framerate = 30

# Stream input type: v4l2 (fpv), appsrc (cv)
input = v4l2

# Stream output type: file (save video), udp (stream video), wcast (wifibroadcast)
output = udp

# Output destination: filename (file output), IP address (udp output), Interface (wcast output)
output_dest = 192.168.10.10
# output_dest = wcast0

# Output port: Port number (eg. 5000) for network destination, Channel for wifibroadcast output (eg. 1)
output_port = 5600
# output_port = 1

# Pipeline Override - This is used to provide a manual pipeline if the auto construction fails o
# pipeline_override = v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! autovideoconvert ! vaapih264enc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.111 port=5000 sync=false

closed 2017-11-12 by fnoop