goodrobots/maverick #487: using "wifi-setup"

terial, 2017-06-19:
Using 1.06 on a fresh image, using "wifi-setup" does not seem to work correctly? Maybe I am doing something wrong but if i look into /etc/network/interfaces i see wlan0 and wman0 both have my networks SSID and the correct psk key but after a reboot using "maverick netinfo" shows that there is no SSID associated for it and the ip address at login is 192.168.XXX.144 (sorry cant remember what digits XXX were but it wasnt in a range my router uses. Am I doing something wrong?

terial, 2017-06-20:
So i created a fresh SD card and on startup ran "wifi-setup" and then checked network info with "maverick netinfo"... i am not sure if a reboot is required but check anyways and also rebooted and checked agian. It shows wman0 as managed and the correct SSID but most of the other information is blank. When maverick starts up is shows the IP as 192.168.1.144 but my router shows no devices under that IP and "iconfig -a" shows no address for wlan0. I checked the generated PSK key and it matches to what it should. Kind of stumped. I also tried to create a localconf.json to setup maverick using "maverick configure --env=flight" but i got errors from my localconf.json file. I am sure I have quiet a bit wrong. I have never really worked with anything beyond shell scripts and config files. I attached my interfaces file after a fresh image write and the localconf.json file i was trying to create previously but gave up on.
[maverick files.zip](https://github.com/fnoop/maverick/files/1086693/maverick.files.zip)

fnoop, 2017-06-20:
@terial So the localconf.json was a little mangled, you had trailing spaces after some of the last attributes in a block, and you lost track of opening brackets.  It's not Maverick being picky btw, that's just JSON for you :) . I've attached a corrected version.  
[terial-localconf.json.zip](https://github.com/fnoop/maverick/files/1088520/terial-localconf.json.zip)

terial, 2017-06-20:
Thanks fnoop! I have a lot to learn when it comes the software side of things... always been more hardware associated but passion drives the need to learn. Anyways I will try the file you attached. Thanks in advance for fixing my mangled file lol. Hopefully I can test and end up helping you out with maverick, There seems to be a ton of potential!

fnoop, 2017-06-28:
@terial Sorry I don't think you get notified for comment edits so I missed this.  No, it should default to 'managed', odd that you're getting access point.  Could you make sure you have no 'mode' set for that interface in localconf.json, run `maverick configure` again, and if you still get the same please post the output of `maverick netinfo`, contents of /etc/network/interfaces?  Thanks :)

fnoop, 2017-06-28:
Move to milestone 1.1.1 to improve wifi-setup and docs.

terial, 2017-07-03:
Sorry for the delayed response. After a fresh image and running the localconf.jason file you cleaned up for me it seems to work. This time it got an IP from my router after a couple minutes from a reboot so maybe it is something on my side? Anyways here is the output of "maverick netinfo"

Interface: lan0
Type:                   Ethernet
MAC Address:            XX:27:eb:71:ab:XX
Vendor:                 RaspberryPi : Standard Microsystems Corp. : Raspberry Pi Foundation
Model:                  SMSC9512/9514 Fast Ethernet Adapter : ec00
Driver:                 smsc95xx
IP Address:             None
Predictable Name:       enxb827eb71abe5

Interface: wcast1
Type:                   Wireless
MAC Address:            XX:c0:ca:91:ef:XX
Vendor:                 Ralink : Ralink Technology, Corp. : ALFA, INC.
Model:                  RT3572 Wireless Adapter : 3572
Driver:                 rt2800usb
IP Address:             None
Predictable Name:       wlx00c0ca91ef59
Wireless Chipset:       Ralink RT2870/3070
Wireless Standards:     a,b,g,n
Wireless Modes:         ibss,managed,AP,AP VLAN,wds,monitor,mesh
Wireless Bands:         Dual Band - 2.4Ghz & 5Ghz
Wireless Mode:          Monitor
Wireless SSID:          None
Wireless Frequency:     5.745 Ghz
Wireless Chanwidth:     None
Wireless RSS:           None

Interface: wman0
Type:                   Wireless
MAC Address:            XX:27:eb:24:fe:XX
Vendor:                 RaspberryPi : None : Raspberry Pi Foundation
Model:                  None : None
Driver:                 None
IP Address:             192.168.1.28
Predictable Name:       wlxb827eb24feb0
Wireless Chipset:       Unknown
Wireless Standards:     b,g,n
Wireless Modes:         ibss,managed,AP,p2p_client,p2p_go,p2p_device
Wireless Bands:         Single Band - 2.4Ghz
Wireless Mode:          Managed
Wireless SSID:          XXXX
Wireless State:         Associated
Wireless Frequency:     2.412 Ghz
Wireless Chanwidth:     HT20
Wireless RSS:           -54

Default Gateway:        192.168.1.1

Here is my /etc/network/interfaces contents
# Interface lo
auto lo
iface lo inet loopback
# Interface eth0
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
# Interface lan0
auto lan0
allow-hotplug lan0
iface lan0 inet dhcp
# Interface wlan0
auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
    wpa-ssid
    wpa-psk

# Interface wman0
auto wman0
allow-hotplug wman0
iface wman0 inet dhcp
    wpa-ssid XXXX
    wpa-psk  XXXX862b7f4a17205cddfe40b59e01d350228151fd615c016d374d2d52b7XXXX

terial, 2017-07-21:
Just an update, tried on 1.1.1 and after reboot using "maverick netinfo" it shows wlan0 being managed and connected to the correct SSID but IP addresss is listed as None. This was a fresh image and using the RPI3 internal wifi interface.

terial, 2017-07-21:
So i noticed that after running configure with my localconf.json, after 4 minutes maverick will get an IP address and be connected to my network. At login i think the IP shows 192.168.1.144 and shows it is connected my my network but no IP address. This was repeatable after every reboot.

fnoop, 2017-07-21:
Interesting, I had this problem as well but thought I fixed it in the latest release.  When you boot, and you don't get an IP address, can you do:
sudo dhclient <int>
And see if you get an immediate IP address?

terial, 2017-07-22:
running dhclient did not work, still 4-5 minutes before maverick gets an IP and is connected to my network. What is strange is I have a second RPI3. i did wifi-setup and then did a reboot. I then edited to localconf.json and noticed my network information was added as the default so i edited that .conf file. When i login maverick shows the correct IP address however it still takes a few minutes before it is connected.

fnoop, 2017-07-22:
Sorry, github swallowed the important bit of that:
`sudo dhclient <wlan0>`
You have to specify the interface after dhclient

fnoop, 2017-07-22:
If you can confirm this is the same problem for you, I'll look at fixing it again..

terial, 2017-07-23:
I tried sudo dhclient wlan0 but it said cnnot find device wlan0

Trying wman0 instead i got this"
[flight] [mav@maverick-raspberry ~]$ sudo dhclient wman0                                                                                                                                                                                                                                      
mv: cannot stat '/etc/samba/dhcp.conf.new': No such file or directory

Tried again and never got any output back.
[flight] [mav@maverick-raspberry ~]$ sudo dhclient wman0

fnoop, 2017-07-23:
The second one is good - you shouldn't get any output back, but it should magically restore your IP to wman0 interface.

terial, 2017-07-23:
that did the trick! Connects to the network within 30 seconds of powering on.

closed 2017-07-23 by terial

fnoop, 2017-07-23:
Have you updated to a recent version?  (at least 1.1.1)
I thought I had fixed this recently:
https://github.com/fnoop/maverick/issues/428

fnoop, 2017-07-23:
Sorry, just seen in earlier comments you've already updated to 1.1.1.  OK I'll reopen the other issue and continue work on it!

terial, 2017-07-23:
Ok, let me know if you need me to test anything. My router is a linksys wrt1900ac running DD-WRT so i would think it should handle dchp leases decently by who knows. FYI i did log into the router control page and delete the link for maverick-raspberry that had an ip of 192.168.1.28. I think this releases the IP address but not 100% sure.