goodrobots/maverick #488: clarification on maverick-visiond.conf for wifibroadcast

terial, 2017-06-21:


fnoop, 2017-06-21:
The output_port is the wifibroadcast channel, which will probably always be 1 unless you're doing funky multi-channel stuff.  The wifi params (frequency) are set in /srv/maverick/data/config/network/monitor-interface-wcast0.conf.

fnoop, 2017-06-21:
Which wifi dongle / chipset are you using?  I may need to add support, the only one it supports currently is rt2800usb.  You can see how it sets up in /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/files/monitor-interface.sh

terial, 2017-06-21:
the wifi device is ALFA AWUS052NH v2 (2.4Ghz/5Ghz, Diversity, high power, Ralink RT3572 chipset)
I know it is supported in EZ-WifiBroadcast and although the source has been updated to match his latest image it is still missing quiet a items to be able to build from scratch. Having a working wifi setup that could potentially work long range is my biggest priority right now. I know I have bombarded you with questions so I am thinking if you need to add support I may try to do a build from the EZ-Wifibroadcast source because it has a few items i like especially the OSD and relay my findings to you... if you think that may help?

fnoop, 2017-06-21:
Cool looks like a nice adapter.  I'm perfectly happy to add support - can you try running:
```
/srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/files/monitor-interface.sh <interface-name>
maverick netinfo
```
See what comes out?

terial, 2017-07-03:
"/srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/files/monitor-interface.sh start wcast1" output this

Starting interface 'wcast1' with chipset rt2800usb, channel
Usage: /sbin/iw [options] dev <devname> set bitrates [legacey-<2.415> <legacey rate in Mbps.*] [ht-mcs-<2.415> <MCS index.*] {vht-mcs-<2.415> <NSS:MCSx,MCSy... | NSS:MCSx,MCSy.*] [sgi-2.4| lgi-2.4] [sgi-5| lgi-5]

Sets up the specified rate masks.
Not passing any arguments would clear the existing mask (if any)
Options:
        --degbug                    enable netlink debugging

fnoop, 2018-06-01:
Network interface setup has been refactored since, including how monitor interfaces are setup.  It will now warn but not reject non-recognised hardware.

closed 2018-06-01 by fnoop