goodrobots/maverick #495: Error on Bootstrap Ubuntu 16.04 MinnowBoard Turbo Dual

cglusky, 2017-06-26:
Seemed like initial bootstrap went fine.  Rebooted and logged in aa mav user.

ran `maverick --env=dev configure` and got this....

`Environment marker set but is being ignored as --env is set
Maverick Environment:    flight 
Proceeding to update system configuration - please be patient, this can take a while..

Warning: The string '7' was automatically coerced to the numerical value 7 at /srv/maverick/software/maverick/manifests/puppet-modules/ssh/manifests/init.pp:283:32
Warning: The string '2' was automatically coerced to the numerical value 2 at /srv/maverick/software/maverick/manifests/puppet-modules/ssh/manifests/init.pp:284:32
Error: Evaluation Error: Unknown variable: '::sshrsakey'. at /srv/maverick/software/maverick/manifests/puppet-modules/ssh/manifests/init.pp:721:14 on node rsi-mbt.local.tld

Maverick finished, happy flying :)
`

fnoop, 2017-06-26:
Wow that's a new one :)
What version of ubuntu?  
What does 'puppet --version' return?

cglusky, 2017-06-26:
Puppet = 4.10.4
Ubuntu = 16.04.2
Kernel = 4.8.0-56-generic

cglusky, 2017-06-26:
The warnings seem simple enough

https://docs.puppet.com/puppet/latest/release_notes.html#automatic-string-to-numeric-conversions-deprecated

No idea on the error.

fnoop, 2017-06-26:
Very odd.  The coerced warnings are brand new to a puppet 4.10.4 showing in a browser near you, but the actual error is a new one on me.  Is this amd64 architecture? (facter |grep arch)

cglusky, 2017-06-26:
Yes.  It's a minnowboard turbot.  Dual core Atom.

fnoop, 2017-06-26:
What does 'facter -p |grep sshrsa' give?  It should give something like this:
```
[dev] [mav@maverick-joule ~/software/maverick]$ facter -p |grep sshrsa
sshrsakey => AAAAB3NzaC1yc2EAAAADAQABAAABAQC60wLLoDTcr/Q72qd/79Wa8yrzDlOF8PBvF4F6X4vtrrPgTbwjGwn1BQGG/gjX+T4MPQE0Yz8vJL4LNEcRVjtMYeo+enB0RJZ+rCfd3fbAVLWyFs/g21Jpq0sHTMnJ1Jvc3Lv6cvExqVlLNo/7oCjTKnTTiyIEp6EmqC0Q5mhb1CZepprcJuTeu+cMkyWX17fMXU2/nxHbyZjanyMJ/ikuLNyfC03KFfOoRlfvJRtshFOLfcwo2u47xz1VkkWdWydowyZ5A4bFY9qDGggbmPXbafpqHGIwnan28n8fJDzV5Mkv7ufKPwa1NDjelLMe6aUz+R66gOGmGCDRXKl+aAZJ
```

fnoop, 2017-06-26:
Also, what does `gem list |grep facter` return?
```
[dev] [mav@maverick-joule ~]$ gem list |grep facter
facter (2.4.6)
```

cglusky, 2017-06-26:
`facter -p|grep sshrsa` returns nothing

`gem list |grep facter` returns facter (2.4.6)

fnoop, 2017-06-26:
Well this is an odd one.  That fact comes from the core facter package, so there's something a bit odd with your system.  Do you have the ssh client installed? (openssh-client) Have you ssh'd in or out of the mav account?
https://tickets.puppetlabs.com/browse/PUP-5058?jql=text%20~%20%22sshrsakey%22
https://github.com/puppetlabs/facter/blob/master/lib/src/facts/resolvers/ssh_resolver.cc

cglusky, 2017-06-26:
Thanks!  Had to:  

`sudo apt-get install ssh`

Ubuntu Desktop does not have server installed by default.  I guess you might want to check for it on initial bootstrap since I am using a vanilla Ubuntu Desktop that many of us noobs would use.

bootstrap is running.  i'll update you when it's done.

cglusky, 2017-06-26:
bah.  gets part way in and i get what looks like a kernel panic.  updated minnowboard firmware to latest and brought out my monster power supply to make sure i have all bases covered.  last time it bumped me out to the console.  looked like a ctl alt f1 kind of action.  reinstalling from scratch and trying again.  i suspect it's the cheap import ssd.  we shall see.

fnoop, 2017-06-26:
Ah sorry if you're logged in through desktop it will kick you out - default is to stop the desktop.
If you want to change, set a localconf parameter (~/data/config/maverick/localconf.json) as described here:
https://fnoop.github.io/maverick/#/modules/desktop

cglusky, 2017-06-27:
OK.  Making sense.  So in addition to the ssh requirement for Ubuntu Desktop so far...

Prior to maverick --env=dev configure

ctl-alt-f1
sudo systemctl disable lightdm.service

cglusky, 2017-06-27:
Last attempt was yielding a bunch of failed dependencies.  Likely due to errors induced from my previous failed attempts.  Giving Ubuntu Desktop one more shot.  If that does not work I'll try vanilla Debian.

fnoop, 2017-06-27:
Not surprised you'll get some errors, to be honest because I've made some of the manifests quite platform specific it's probably not as 'genericy' as it should be.  If you could pastebin or attach here the complete out of the run (you should find log files in ~/var/log/maverick), that should tell me enough to fix the problems.  Thanks for testing and the feedback :)

cglusky, 2017-06-27:
Here's the log.  Appears I did not get very far before failure.  Running on a wifi connection that depnds on ath9k drivers.  Works fine out of box with Ubuntu.  I am aware of the config files you talk about in your docs.  Happy to dig through in more detail and change things as required.  Just wanted to try it out using your QuickStart.  I know this is in early stages so happy to help out from perspective of a user who knows just enough linux to be dangerous.  That is why I am here.  

[maverick.dev.2017-06-27-04-57.txt](https://github.com/fnoop/maverick/files/1105246/maverick.dev.2017-06-27-04-57.txt)

fnoop, 2017-06-27:
Looks like you had a bunch of package failures, but it was installing earlier.  One thing I've noticed is that Ubuntu/Debian tends to randomly kick off apt/dpkg processes/updates in the background which can interfere with anything else trying to use the package manager.
Could you please try running it again, and posting the output?  That will determine which of the errors are actual errors and not just transient.
The amount of errors/warnings you see is because of the dependency graph, one error tends to everything else out as well by design.

cglusky, 2017-06-27:
I have seen the same thing with the dpkg process.  Files being locked etc.  Annoying!  Is that on a cron or something that can be killed?

Running it again.

fnoop, 2017-06-28:
Very good point, been meaning to do something about this for a while!
https://github.com/fnoop/maverick/issues/502

cglusky, 2017-06-28:
log from last run.  there was a more helpful error in the console.  sorry, can't remember exact verbiage and i do not see it in log file.  it was during the db install.  something about forcing install with noauth flag without noauth enabled?

[maverick.dev.2017-06-27-16-23.log.txt](https://github.com/fnoop/maverick/files/1108961/maverick.dev.2017-06-27-16-23.log.txt)

fnoop, 2017-06-28:
https://github.com/fnoop/maverick/issues/504

fnoop, 2017-06-28:
@cglusky I can't actually see what's going wrong because of the missing logs.  Could you please self-update and run again?  I've fixed the logging (hopefully) in #504

fnoop, 2017-06-28:
Just to confirm, are you running master branch?  That's where I'm making these changes.  Set ~/data/config/maverick/maverick-branch to:
```
MAVERICK_BRANCH=master
```
Then run `maverick self-update` and you should see it unshallow the repo and switch branches.

cglusky, 2017-06-29:
Changed to master branch.  Ran self-update.  Log attached.  Seems to fail on trying to start maverick-influxdb.  I confirmed influxdb is installed.  I can even start it with `sudo service influxdb start` 

[maverick.dev.2017-06-29-01-56 .log.txt](https://github.com/fnoop/maverick/files/1110644/maverick.dev.2017-06-29-01-56.log.txt)

fnoop, 2017-06-29:
Thanks, that log file is now much easier to read!
OK so there's a dependency issue - it's defining the maverick-influxd service:
```
Notice: /Stage[main]/Maverick_analysis::Influx/File[/etc/systemd/system/maverick-influxd.service]/ensure: defined content as '{md5}a5fd16d60a739ecdd91afa5d1dd0680e'
Notice: /Stage[main]/Maverick_analysis::Grafana/File[/srv/maverick/var/log/analysis/grafana]/mode: mode changed '0750' to '0755'
Notice: /Stage[main]/Maverick_vision::Visiond/Service_wrapper[maverick-visiond]/Service[maverick-visiond]/ensure: ensure changed 'stopped' to 'running'
Error: Could not enable maverick-influxd:
Error: /Stage[main]/Maverick_analysis::Influx/Service_wrapper[maverick-influxd]/Service[maverick-influxd]/ensure: change from stopped to running failed: Could not enable maver
ick-influxd:
```
Which should notify systemd to reload in order to pick up the new service manifest, but it seems to be trying to start the service as a dependency to this reload, which is the wrong way round:
```
Notice: /Stage[main]/Base::Functions/Exec[maverick-systemctl-daemon-reload]: Dependency Service[maverick-influxd] has failures: true
Warning: /Stage[main]/Base::Functions/Exec[maverick-systemctl-daemon-reload]: Skipping because of failed dependencies
```
I'll try and work out how to resolve this.  In the meantime, you can carry on by just running `maverick configure` again - it wouldn't hurt to do `sudo systemctl daemon-reload` beforehand just to ensure the new systemd manifest is loaded.

Also note that the service 'influxdb' is deliberately stopped and disabled, and a different 'maverick-influxd' service defined.  Almost all services that run under Maverick control have a separate service manifest customised to running under a single 'mav' user and relocating data and sometimes binaries.  This also means that all maverick services are prefixed by 'maverick-' and are very easy to differentiate in the noise of systemd.

fnoop, 2017-06-29:
Also this error is odd:
```
Error: /Stage[bootstrap]/Base::Maverick/File[/srv/maverick/.cache]: Failed to generate additional resources using 'eval_generate': Cannot manage files of type socket
```
This probably means you have a socket file hidden underneath /srv/maverick/.cache, could you do:
`find ~/.cache -type s` and output what the socket is please?

cglusky, 2017-06-29:
Getting closer.  Just one error on try to create the database...

[maverick.dev.2017-06-29-13-48.log.txt](https://github.com/fnoop/maverick/files/1112101/maverick.dev.2017-06-29-13-48.log.txt)

cglusky, 2017-06-29:
mav@mbt:~$ service maverick-influxd status
● maverick-influxd.service - InfluxDB is an open-source, distributed, time serie
   Loaded: loaded (/etc/systemd/system/maverick-influxd.service; enabled; vendor
   Active: inactive (dead) (Result: exit-code) since Thu 2017-06-29 13:49:49 BST
     Docs: man:influxd(1)
 Main PID: 32521 (code=exited, status=1/FAILURE)

cglusky, 2017-06-29:
Rebooted...

mav@mbt:~$ service maverick-influxd status
● maverick-influxd.service - InfluxDB is an open-source, distributed, time serie
   Loaded: loaded (/etc/systemd/system/maverick-influxd.service; enabled; vendor
   Active: active (running) since Thu 2017-06-29 14:07:45 BST; 8min ago
     Docs: man:influxd(1)
 Main PID: 830 (influxd)
   CGroup: /system.slice/maverick-influxd.service
           └─830 /usr/bin/influxd -config /srv/maverick/data/config/analysis/inf

ran again and did not get any errors.

[maverick.dev.2017-06-29-14-10.log.txt](https://github.com/fnoop/maverick/files/1112152/maverick.dev.2017-06-29-14-10.log.txt)

is there a way to check status of the system.  like a stop light chart for everything installed that is supposed to be running?

fnoop, 2017-06-29:
https://fnoop.github.io/maverick/#/?id=statusinfo
The 'ooh pretty graphs' have a system dashboard as well to monitor the local system.

cglusky, 2017-06-29:
I am going to start over and make sure it works from scratch.  I am not convinced everything compiled along the way.  More soon.  Thanks!

cglusky, 2017-07-04:
Started fresh.  Moved to ethernet connection since wireless in my shop is a two hop adhoc mess.

Not there yet sadly.  Log attached.  This is the error I was trying to remember earlier...

> There were unauthenticated packages and -y was used without --allow-unauthenticated

[maverick.dev.2017-07-03-19-02.log.txt](https://github.com/fnoop/maverick/files/1122893/maverick.dev.2017-07-03-19-02.log.txt)

fnoop, 2017-07-08:
That's an odd error - the key has already been input, perhaps it's because an apt update hasn't happened yet before it tries to install the package.  I'm just working out the last details of release 1.1.1 and I'll try and address this.  Was that running the master branch?  I thought I pushed a fix for the coerced warnings.

cglusky, 2017-07-08:
Yep. Master.

fnoop, 2017-07-09:
Oh goody, puppet happened to do a major release to 5.0.0 since.  I'll create another issue to track and support that.

fnoop, 2017-07-09:
@cglusky Could you please make sure you're on master branch, and do self-update and configure again please?

cglusky, 2017-07-10:
Will do!  Spotty network right now.  Might be a few days.  Have the MinnowBoard mounted to new dev frame design (the bigger one) so really interested to get this working.

fnoop, 2017-08-05:
@cglusky If you have a few minutes, could you try again with 1.1.2 (just pushed to stable)?

cglusky, 2017-08-06:
Sorry for delay.  Will give it another try and reports back as soon as I can.  Hopefully tomorrow.

cglusky, 2017-08-10:
No luck.  First try after a good =bootstrap the wifi hardware was not happy.  Reboot and got connected and got further on =dev but it seemed to fail on the ardupilot git sequence.  Also, still having issues with something locking dpkg.  This was after =bootstrap and I had even turned off auto updates via GUI.  Took a few reboots to clear the lock.  At least your new code is checking for the lock first.  Sad to say that Ubuntu update processes are starting to feel Windows Creepy.

[maverick.dev.2017-08-10-00-24.log.txt](https://github.com/fnoop/maverick/files/1215443/maverick.dev.2017-08-10-00-24.log.txt)

fnoop, 2017-08-14:
Two issues in the log:
 - Network stops working
 - Takes too long to compile collectd so times out (fixed in latest stable)

But most of the run does work - in a CM (configuration management) system like this if one small thing fails then you'll se a whole bunch of other dependent resources fail too.  It looks dramatic but it's really not, you usually just have to fix the one small thing and everything else will clear and carry on.  In general, try running the configure again once or twice, it often clears any transient errors.

cglusky, 2017-08-15:
I'll run it a few more times.  Thanks for the help!

cglusky, 2017-08-15:
Getting closer.  Still having issues with ardupilot and/or waf...

[maverick.dev.2017-08-15-20-53.log.txt](https://github.com/fnoop/maverick/files/1226276/maverick.dev.2017-08-15-20-53.log.txt)

fnoop, 2017-08-15:
Getting close!!  
The ardupilot clone submodules partially failed due to network issues so you have a damaged ~/code/ardupilot area.  You could repair it (git submodule init; git submodule update), but probably easier just to remove it and let maverick start again:
```
rm -rf ~/code/ardupilot
maverick configure
```

cglusky, 2017-08-15:
That did it.  looks like we might have a bootstrapped minnowboard!

status shows all sitl services running.

any tests/steps you want me to perform before trying to make an iso?

cglusky, 2017-08-15:
collectd is not coming up on startup.  a systemctl restart maverick-collectd brings it back up fine.

cglusky, 2017-08-15:
adding a note so i don't forget.  ttyS0 is the default serial on a minnowboard.  hooked to pixracer for this install.  just need to make sure config for mavrouter is correct.

fnoop, 2017-09-25:
@cglusky If you get a minute, could you pull master and see if it still runs OK on the minnowboard?  I've done some work on services starting on slower systems, so you collectd on boot issue hopefully should be fixed.  I'm looking to make 1.1.4 as a 'platform' release.

rocketshipsys, 2017-09-27:
will do.  taking a short break from the shop.  will try to have it done by end of next week.

fnoop, 2017-10-23:
@cglusky Going to assume this is working on the minnowboard now.  Any more problems, please do reopen this.

closed 2017-10-23 by fnoop