goodrobots/maverick #509: Old OS images for Odroid

Cosmonaut65, 2017-07-02:
It appears the images listed at https://fnoop.github.io/maverick/#/?id=os-images are all version 1.0.6 except for Odroid which is at 1.0.4.

Is there anywhere to download an image for 1.0.6 for Odroid?
Alternatively, would using the Bootstrap method get me the latest version? I'm particularly after the tensorflow library.

fnoop, 2017-07-02:
@Cosmonaut65 Theoretically you should be able to just do:
```
maverick self-update
maverick configure
```
However, tensorflow was particularly difficult to build on ARM and I believe there might be extra steps for the odroid, rather than the raspberry build which is ready to go.  If you have patience to help debug the build if it goes wrong then I'm happy to fix it (@patrickpoirier51 might be able to help), unfortunately my odroid died a few weeks ago so I don't have one at the moment.

patrickpoirier51, 2017-07-02:
@Cosmonaut65 
 I builded Tensor flow on my Odroid a while ago.
Look here:
 https://github.com/samjabrahams/tensorflow-on-raspberry-pi/issues/41#issuecomment-249463027

closed 2018-01-22 by fnoop