goodrobots/maverick #522: maverick configure fails on network configuration

kiranshila, 2017-07-12:
maverick_network is enabled, maverick_network::interfaces is copied from example, I get the following error:
`Error: Evaluation Error: Error while evaluating a Resource Statement, Evaluation Error: Unknown variable: 'netmask' . at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/interface_managed.pp:94:32 at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/process_interface.pp:44 on node odroid`

kiranshila, 2017-07-12:
Also, this error doesn't seem to occur when running the master branch

fnoop, 2017-07-12:
Thanks I hit this too a while back, thought I fixed it but looking at the code it's definitely missing the parameter so I've added it for good measure.

closed 2017-07-12 by fnoop