goodrobots/maverick #539: Odroid XU3/XU4 MFC decoding

maartenvds, 2017-07-18:
First and all, sorry for hijacking this issue platform, but I saw that you had a couple of issue tickets related to MFC hardware video decoding on an odroid board ([#218](https://github.com/fnoop/maverick/issues/218) and [#156](https://github.com/fnoop/maverick/issues/156)), so I thought you might be able to help me out. I'm trying to decode an RTSP video stream in real time on an odroid XU3/XU4 board using the MFC hardware.
I installed the 'official' ubuntu 16.04 odroid image with the latest 4.9 kernel which should have support for the MFC. I got it to work using gstreamer compiled according to these [instructions]( https://forum.odroid.com/viewtopic.php?f=95&t=24366). The problem is that its extremely slow. I have posted some measurements over [here](https://forum.odroid.com/viewtopic.php?f=95&t=27471#p195176). With a simple pipeline (hardware decoding and software colorspace conversion) I only get 4 fps (30fps 1080p stream) and its still claiming roughly 100% CPU... FFmpeg can easily decode it at 30fps using like 80% CPU but without the hardware decoder (I assume). I guess there is something seriously wrong with my setup, but I'm kinda clueless...

I can hardly find any benchmarks for the MFC that contain detailed setup instructions, so I was wondering what your experience is with this. Are you able to get something useful out of it and if so, what decoding/encoding rates do you get and what did you do to accomplish them? 

Thanks in advance, and again sorry for this non related issue

fnoop, 2017-07-21:
I'm not sure I ever actually tried decoding, I was more focused on encoding, but it works the same way, using the same mechanisms and software so it should work.   Sounds like you're doing stuff in software somehow, what pipeline are you using?

fnoop, 2017-07-21:
Have you looked at how Maverick constructs gstreamer for odroid?
https://github.com/fnoop/maverick/blob/stable/manifests/maverick-modules/maverick_vision/manifests/gstreamer.pp#L71
https://github.com/fnoop/maverick/blob/stable/manifests/maverick-modules/maverick_vision/manifests/gstreamer.pp#L40

It uses a patched gstreamer tree in my github, so you would need to set:
`"maverick_vision::gstreamer::gstreamer_version": 1.10.2`

Or you could just try the maverick odroid image, it's a bit old but I think it included the hardware MFC support.  Doesn't the latest odroid experimental kernel include the patched gstreamer?

fnoop, 2017-07-21:
Sorry, just read your posts on the odroid forum.  I see you've got basic hw encoder working -
 that's good.  Unfortunately I don't think I ever tested the decoder - I was only interested in encoding video to transmit over wifi from UAV.  Have you tried a simple decode from h264 to screen or file, just to make sure the decoder device is correct and utilised?
I note you're using v4l2video10videodec because there's no v4l2video10h264dec.  I wonder if the patched gstreamer doesn't include h264 decoder support yet?

maartenvds, 2017-07-21:
Thanks for your input, I'll give the maverick odroid image a try next week to see what it does.
I think the patched gst implements the hardware decoder since a dmesg | grep mfc produces the following message  every time I invoke a pipeline containing (only) the hardware decoder:

s5p-mfc 11000000.codec:: vidioc_s_fmt queue busy

Also if I put strace in front of the gst-launch command, it prints messages about opening the /dev/video10 device which in my case is the hardware decoder. This is by the way not the case with patched ffmpeg (which apparently only implements the encoder). However I cannot recall that I actually looked at the output of a decoded peace of video by a gst pipeline to see if its correct, have to check...

fnoop, 2017-07-21:
It's unlikely you'll get anything more by running maverick odriod image - I think the experimental kernel probably has at least the same support.  I swapped my odroid out because I got fed up with waiting for decent hardware support.  I'll close this out but please do feel free to ask any more questions here, if I can be of any help I'll try :)

closed 2017-07-21 by fnoop