goodrobots/maverick #540: Ability to download logs with Grafana

terial, 2017-07-20:
Would it be possible to get the ability to download logs through Grafana/web browser. Also it would be handy to be able to delete the logs after they start accumulating.

fnoop, 2017-07-20:
Good idea!
- To download the logs, I can add links to the grafana index dashboards, and hopefully symlink through the archive directories to the grafana webserver.  Maverick doesn't have a webserver yet, that's on the todo list (for the upcoming web interface).

fnoop, 2017-07-20:
To delete the dataflash logs, that's a bit more tricky.  We could add a 'housekeeping' process that deletes all logs older than x days (disabled by default?).  People might get upset if they were disappearing by default.  Another option would be to automatically compress when archiving.

Separately, there should probably be some housekeeping on the influxdb which will end up taking a lot more space, in particular the system logs.  A default retention period of a week or so might be sensible as they are really intended for debugging/correlation when developing companion computer systems, eg. to see the effect vision apps have on the system.

terial, 2017-07-21:
The house keeping idea is great! If you could set a keep since time that would be awesome. Grafana seems like a great addition to Maverick and I agree about the system logs. I use my UAV for surveying so managing logs and data at the moment is very tedious. Once I saw your implementation of grafana ideas started popping up lol. I like the system logs for exactly the reason you described, it's nice to see how much resources are being used but for everyday use may not need to be logged. Is there a way a real time mode could be monitored. It won't get logged, well maybe just long enough to be graphed for a few minutes but isn't saved? I guess this would really be a log that goes back only 5 minutes?

fnoop, 2017-07-31:
Added system data housekeeping, to just keep the past 3 days.
https://github.com/fnoop/maverick/issues/545

fnoop, 2017-08-01:
Ok the builtin webserver to grafana doesn't allow symlinks or to alter the routing, so this isn't easily achievable without some significant hacks.  Therefore the ability to download logs through browser will have to wait until a proper webserver is put into place (which is planned). 
Until then, you can sftp/scp the logs from mav@<hostname>:data/mavlink/archive

closed 2017-08-01 by fnoop