goodrobots/maverick #543: Simplify initial configuration on fresh install

terial, 2017-07-21:
I like that when I do a fresh raspbian image on raspberry pi I can put and empty file named "ssh" and "wpa_supplicant" with the wpa_supplicant file containing my network info on /boot so when i do the first powerup I can ssh over my network and continue to work on the raspberry. Would it be possible if a localconf.json file was already created to put this on /boot along with an empty file named something like "install_conf=flight" that would let maverick know to copy the original localconf.json and rename to localconf.json.bak and that I have provided a new localconf.json file to copy over and configure with the environment set to flight and then reboot. This seems like an easy way to power on, wait a few minutes (as long as no major changes are required i suppose) and then connect over ssh or the cloud9 interface with a web browser? I understand everyone's has different needs so my logic may not justify the work to implement something like this, if so its still not too much work to unplug my monitor and swap a USB receiver for my keyboard back and forth :)

fnoop, 2017-07-21:
Yes, this is a good idea.  I find having to create the empty ssh intensely irritating - I wish it would just load sshd by default.  I assume they wanted to stop putting millions of potential raspberry zombies on the net..

terial, 2017-07-22:
Can I ask how this would be implemented? I can work with shell scripts decently but I have not got myself into python which maverick seems to take advantage of. I can try to get a rough shell script implemented if it helps you but it it will be done with python I am not sure how beneficial a shell script will be to you...

fnoop, 2017-07-22:
There's already a similar mechanism that copies node configs in during bootstrap (which is in pretty much the only shell code in Maverick!):
https://github.com/fnoop/maverick/blob/stable/bin/maverick#L613

As it happens the population/handling of localconf.json is just about to change in another issue:
https://github.com/fnoop/maverick/issues/537#issuecomment-315971526

So I'll add this change at the same time.

fnoop, 2017-08-05:
Maverick already copies local-node files in from bootstrap location:
https://github.com/fnoop/maverick/blob/master/bin/maverick#L618

Add similar very simple mechanism to copy/move file hierarchy from /boot or other similar location depending on platform and use for both bootstrap and inplace maverick configure.  Also need to add a boot action to check for existence of such hierarchy and if present then to trigger maverick configure - this will allow headless setup such as suggested by @terial.

fnoop, 2017-08-05:
In order to prevent this mechanism from doing anything nefarious to the filesystem/maverick or other components, this should be restricted to config only, so /srv/maverick/data/config.

fnoop, 2017-08-05:
This isn't going to easily work with a bootstrap, it's real usefulness is for configuring already bootstrapped images on headless platforms like raspberry/joule etc.  So it just needs to work for final destination /srv/maverick, not transient bootstrap location.

fnoop, 2017-08-05:
Moved ~/.environment to ~/data/config/maverick/maverick-environment.conf so it can be part of this mechanism: #558

fnoop, 2017-08-05:
Added facility to pick up config in /boot/maverick and move into place.  Also added systemd manifest to detect /boot/maverick on boot and call maverick configure automatically if exists.

closed 2017-08-05 by fnoop