goodrobots/maverick #544: Error in config for mavlink-router

terial, 2017-07-21:
When trying to start mavlink-router@fc maverick status always shows status as stopped. When looking at the maverick log for the service it shows an error:

Displaying service log: mavlink-router@fc
-- Logs begin at Fri 2017-07-21 06:55:42 BST. --
Jul 21 06:56:58 maverick-raspberry systemd[1]: Starting Maverick mavlink-router Mavlink Proxy - fc...
Jul 21 06:56:58 maverick-raspberry systemd[1]: Started Maverick mavlink-router Mavlink Proxy - fc.
Jul 21 06:56:58 maverick-raspberry systemd[1]: maverick-mavlink-router@fc.service: main process exited, code=exited, status=1/FAILURE
Jul 21 06:56:58 maverick-raspberry systemd[1]: Unit maverick-mavlink-router@fc.service entered failed state.
Jul 21 06:56:58 maverick-raspberry mavlink-routerd[1569]: On file /srv/maverick/data/config/mavlink/mavlink-router-fc.conf: Line 14: Invalid value '115200,115200,57600' for field 'baud'
Jul 21 06:58:01 maverick-raspberry systemd[1]: Starting Maverick mavlink-router Mavlink Proxy - fc...
Jul 21 06:58:01 maverick-raspberry systemd[1]: Started Maverick mavlink-router Mavlink Proxy - fc.
Jul 21 06:58:01 maverick-raspberry mavlink-routerd[1856]: On file /srv/maverick/data/config/mavlink/mavlink-router-fc.conf: Line 14: Invalid value '115200,115200,57600' for field 'baud'
Jul 21 06:58:01 maverick-raspberry systemd[1]: maverick-mavlink-router@fc.service: main process exited, code=exited, status=1/FAILURE
Jul 21 06:58:01 maverick-raspberry systemd[1]: Unit maverick-mavlink-router@fc.service entered failed state.

/srv/maverick/data/config/mavlink/mavlink-router-fc.conf

[UartEndpoint serialin]
    Device=/dev/ttyAMA0
    Baud=115200,115200,57600

It looks like the configuration messes something up. I have my baudrate set at 921600.

fnoop, 2017-07-21:
Couple of things here:
--- 
Baud=115200,115200,57600 means you haven't set your baud rate in localconf as described here:
https://fnoop.github.io/maverick/#/modules/fc?id=fc-mavlink-proxy
You should set an entry like:  
`"maverick_fc::mavlink_baud": 921600`
You should end up with an entry like: Baud=921600,115200,57600
It always sets 115200,57600 as fallbacks.
---
/dev/ttyS0 is not the correct way to access the uart on the raspberry, it should be /dev/ttyAMA0 otherwise you'll hit problems.  Maverick disables bluetooth if necessary and takes back AMA0 for the serial port.
---
Hopefully this will

terial, 2017-07-22:
Thanks I always seem to get the serial port mixed up since the RPI3/bluetooth changed from RPI2. Just for my understanding, Bluetooth can be disabled in config.txt on/boot correct? If so does enable_uart=1 still have be be set in config.txt as well?

I checked my localconf.json and this is what I had (I did change the input from ttyS0 to ttyAMA0:

// Maverick Flight Controller Modules
	// FC Mavlink Proxy
	"maverick_fc::mavlink_input": "/dev/ttyAMA0", // Serial/USB port for communication to flight controller
	"maverick_fc::mavlink_baud": "921600", // Baud rate for communication to flight controller
	"maverick_fc::mavlink_proxy": "mavlink-router", // Current supported proxies are: mavlink-router (the default), mavproxy and cmavnode

I am going to flash again and double check everything.

fnoop, 2017-07-22:
Not sure if it makes a difference, but baud is a numeric so shouldn't be in quotes:
`"maverick_fc::mavlink_baud": 921600`

terial, 2017-07-23:
Thanks fnoop, I know I seem to always be bugging you with a sorts of questions and mistakes and you just keep on coming back with help and answers... you really are a great resource! I see there are some .bin files being saved to the /fc folder but they are not being moved to the inbox for Grafana to see but I will look into that later. One more question and this one has been bugging me for some time with both mavlink-router and cmavnode. Its my understanding that when you have a udpendpoint set in the .conf and the mode is eavesdropping no IP is specified (0.0.0.0) just the port, with maverick using 14573-14575 for udp, that the stream out is not specified to a single target until one is connected. I have tried with mission planner to connect to 14573 thru 14575 but never get a connection but if i specify a specific IP I can connect. The documentation on mavlink-router's github wasnt very clear to me. Am I missing something on this?

fnoop, 2017-07-23:
Ah to be honest I've not actually had time to test the mavlink-router dataflash log files, so if the bin files aren't being picked up then I'll raise that as a separate issue.  They shouldn't be moved to the inbox, but the fc directory itself is watched for processing (once processed it will be moved to the archive directory).

Re: the 'eavesdropping' or udp server ports, in missino planner are you connecting using the 'UDPCI' connection type?

terial, 2017-07-23:
Ah i wondered what UDPCI was for but have never seen anything about it. Just changed some settings back and restarted mavlink-router. Seems like UDPCI is kind of backwards just need to specify the host IP. Works like a charm on all 3 ports specified. Im going to close  this issue.

closed 2017-07-23 by terial

fnoop, 2017-07-23:
Cool!