goodrobots/maverick #546: fc/sitl dataflash logs not processed by mavlogd

fnoop, 2017-07-23:
.bin files created by mavlink-router not being picked up by mavlogd for processing

terial, 2017-07-23:
Not sure if this is related but I have noticed in flight environment "maverick status" shows mavlogd enabled at boot but is not running. running "maverick start mavlogd" starts the service and "maverick log mavlogd" reports this:

Displaying service log: mavlogd
-- Logs begin at Sun 2017-07-23 09:17:01 BST. --
Jul 23 09:17:27 maverick-raspberry systemd[1]: Starting Maverick Log Import Daemon...
Jul 23 09:17:27 maverick-raspberry systemd[1]: Started Maverick Log Import Daemon.
Jul 23 09:17:31 maverick-raspberry systemd[1]: maverick-mavlogd.service: main process exited, code=exited, status=1/FAILURE
Jul 23 09:17:33 maverick-raspberry maverick-mavlogd[1279]: Error: PID file '/srv/maverick/var/run/maverick-mavlogd.pid' not locked
Jul 23 09:17:33 maverick-raspberry maverick-mavlogd[1279]: - mavlogd not running, cannot stop
Jul 23 09:17:33 maverick-raspberry systemd[1]: Unit maverick-mavlogd.service entered failed state.
Jul 23 21:00:50 maverick-raspberry systemd[1]: Starting Maverick Log Import Daemon...
Jul 23 21:00:50 maverick-raspberry systemd[1]: Started Maverick Log Import Daemon.

Looks like maverick is assigning a PID to the process but it is not locked? Decided to reboot and see what the log was before getting mavlogd was started and got this:

Displaying service log: mavlogd
-- Logs begin at Sun 2017-07-23 09:17:01 BST. --
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: func(self)
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: File "/usr/lib/pymodules/python2.7/daemon/runner.py", line 151, in _stop
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: if is_pidfile_stale(self.pidfile):
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: File "/usr/lib/pymodules/python2.7/daemon/runner.py", line 220, in is_pidfile_stale
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: pidfile_pid = pidfile.read_pid()
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: File "/usr/lib/pymodules/python2.7/daemon/pidlockfile.py", line 48, in read_pid
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: result = read_pid_from_pidfile(self.path)
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: File "/usr/lib/pymodules/python2.7/daemon/pidlockfile.py", line 146, in read_pid_from_pidfile
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: "PID file %(pidfile_path)r contents invalid" % vars())
Jul 23 21:08:22 maverick-raspberry maverick-mavlogd[1385]: daemon.pidlockfile.PIDFileParseError: PID file '/srv/maverick/var/run/maverick-mavlogd.pid' contents invalid

I had to google a bit about PID and lock files but is it possible that during an improper shutdown... such as just disconnecting the power which seems like a likely scenario when maverick is used in the field the last PID and lock file remain from a previous session prevention the new PID written to the correct files until a user manually starts it? I checked and both the pid and lock file were empty and now when trying to start the service i get the same output as above. Removing the maverick-mavlogd.pid and lock files would allow the process to start.

fnoop, 2017-07-27:
Lots of work done on collectd/influxd in a series of commits.  Verified now that FC .bin logs automatically picked up, processed and archive.  All data imported into new influx RP (retention policy) 'flightdata' with no expiry.

fnoop, 2017-07-27:
Yes the daemon is rather crude - it's a very old python daemon implementation that leaves something to be desired..  I'll raise a separate issue for that, it may be that under systemd we no longer need a daemon.

closed 2017-07-27 by fnoop