goodrobots/maverick #547: Arducopter SITL running while enviroment is set to flight

terial, 2017-07-23:
Just curious, I noticed a process "arducopter" running that was using more resources than the other process, sometimes going up to 35% while everything else is really low.

mav       1002 16.2  0.3   4812  3724 ?        S    09:05   0:38 /srv/maverick/code/ardupilot/build/sitl/bin/arducopter -S -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults /srv/maverick/code/ardupilot/Tools/autotest/default_params/copter.parm

Looks like something related to SITL? Should SITL items be running when environment is set to flight?

fnoop, 2017-07-23:
Yes this is the SITL, it shouldn't be running in flight env!

fnoop, 2017-07-23:
You're a talented bug finder :)

fnoop, 2017-07-23:
(and clearly there are a lot of bugs!)

terial, 2017-07-23:
I am glad you find it helpful! I have always been a mechanical guy so all this software stuff is fairly new to me which makes me a bit hesitant to raise issues... All these project related to ardupilot and using linux sparked a new learning path. I guess I am learning a few things lol.

fnoop, 2017-07-24:
So I've just tested this, going from dev environment with all the sitl services running, then doing:
`maverick --env=flight configure`
When it's finished, all SITL services are stopped as expected (ps -ef |grep sitl).  Could you confirm how you're getting to flight mode?

terial, 2017-07-25:
I am copying the attached localconf.json onto the /boot directory and then copying he original as localconf.json.bak and then copying the localconf.json from /boot to ~/software/maverick/conf/ and then running maverick --env=flight configure. Everything shows that I am in running the flight environment but I can be good at breaking things. I wont be able to tonight... wife's birthday :) but I will do another fresh image and try it again once you review the attached localconf.json 

[terial localconf.zip](https://github.com/fnoop/maverick/files/1171740/terial.localconf.zip)

fnoop, 2017-07-25:
Your localconf looks good.  Can you try going to dev environment, then back to flight?
```
maverick --env=dev configure
maverick --env=flight configure
```
And attach the logs?  You'll find them in ~/var/log/maverick

fnoop, 2017-07-25:
ps - there's no need to do fresh image - the way Maverick works is that it effectively reconfigures your entire system every time it runs, so you don't get any benefit from reinstalling it (theoretically).

fnoop, 2017-07-27:
Ok I've checked this on several systems.  The only time that SITL processes will remain running in flight mode is if the configure fails for some reason in flight mode, or going to flight mode.  There's not really any way to improve this - maverick configure must run cleanly to provide the intended environment.

closed 2017-07-27 by fnoop

terial, 2017-07-27:
ok so I should be able to reconfigure and then check again to see if any SITL process are running and if so check the log for when maverick was running the configuration process... Thanks for the info!

flouwers, 2018-03-27:
Hello. I'm having the same issue.
I tried switching from flight to dev and vice-versa several times, but the SITL and mavproxy@sitl processes are still active in flight environment.
The configure commands reports no error.
[maverick.dev.2018-03-27-11-00.log.zip](https://github.com/goodrobots/maverick/files/1851601/maverick.dev.2018-03-27-11-00.log.zip)
[maverick.flight.2018-03-27-11-41.log.zip](https://github.com/goodrobots/maverick/files/1851602/maverick.flight.2018-03-27-11-41.log.zip)
[localconf.zip](https://github.com/goodrobots/maverick/files/1851603/localconf.zip)

fnoop, 2018-03-27:
@flouwers thanks for the report and logs.  What platform are you running?

flouwers, 2018-03-27:
Hi @fnoop
I'm running on Raspberry Pi 3

fnoop, 2018-03-27:
@flouwers So the cause of your problem is the same as https://github.com/goodrobots/maverick/issues/607.  If you look in ~/software/maverick/conf/environments/flight/defaults.json, you'll see lines like this at the top:
```
	"maverick_dev::sitl::sitl_active": false, // Whether SITL is active or not
	"maverick_dev::sitl::mavlink_active": false, // Whether mavlink proxy is active or not
	"maverick_dev::sitl::rosmaster_active": false, // Whether ROS master for sitl is active or not
	"maverick_dev::sitl::mavros_active": false, // Whether Mavros for sitl is active or not - requires sitl rosmaster to be active
```
The maverick (puppet) config is hierarchical and is merged across several levels of config defined in ~/software/maverick/conf/hiera.yaml, where the top levels have a higher priority than the lower levels.  localconf.json has the highest priority so anything in there will overwrite anything else at a lower level, and you have these two settings in localconf:
```
"maverick_dev::sitl::sitl_active": true, // Whether SITL is active or not
"maverick_dev::sitl::mavlink_active": true, // Whether mavlink proxy is active or not
```
These are forcing the sitl to turn on.  It's a bit confusing, agreed, but it comes from the power and flexibility of the config system.  I can't just 'turn off' maverick_dev altogether because then the system doesn't know anything about the resources and doesn't know to turn them off.
Because this has caused a bit of confusion i'll keep it in mind and at least make the docs clearer, and if I think of a way to refactor the config logic to make it clearer in the future then I will.

fnoop, 2018-03-27:
btw those two settings are on by default in dev mode so if you just comment them back out, you'll get the behaviour you expect: on in dev and off in flight.