goodrobots/maverick #566: Install issues RP3

lucidwan, 2017-08-14:
Not sure how to report this, figure this is good a place as any.
I've spent a about a week messing with getting install on Pi3 and Odroid XU4.

Here are the issues:
Image writes fine on both. Both boot fine.
PI (2 and 3 actually)-Solution - don't use self-update first. In my case, I setup up wifi, then  wman0 and wap0, in localconf- ran maverick configure, and everything updated correctly.

If you go through 'maverick self-update first' after boot it breaks the system - obviously since we're putting multiple new things and changes in. The problem is, that when you run 'maverick configure' it errors.
The first error is invalid class "nginx".  The modules folder has an empty folder for Nginx. 
This was simple enough to figure out once I figured out puppet and manifests, and the install path for modules using puppet. 
The issue with nginx is that it won't install the correct dependencies - from puppet, that I could find using puppet-nginx and jfryman-nginx. Jfryman throws a server::resource error. Installing puppet-nginx --force fixes it (adds the declaration to the .conf) 
So when you maverick configure, it throws out a ton of dependency errors.
I'm running fine - atm - without using self-update. I'll test this in a little bit.

I'll post a separate one for Odroid.

fnoop, 2017-08-14:
Yup really sorry that was a stupid accidental commit of the dev branch while my brain was still on holiday.  I'm just fixing that now, hopefully this evening.

fnoop, 2017-08-14:
puppet-nginx dependencies are now fixed btw:
https://github.com/voxpupuli/puppet-nginx/issues/1107

I screwed up the module commit.

fnoop, 2017-08-14:
Could you please try:
```
maverick self-update
maverick configure
```
And see if the errors are gone?

lucidwan, 2017-08-14:
Thanks! Sorry I didn't search it, just wanted to get it out there while it was fresh on my mind.

lucidwan, 2017-08-14:
I will! Let you know soon.

lucidwan, 2017-08-14:
Ok:
Error: Evaluation Error: Error while evaluating a Resource Statement, Evaluation Error: Error while evaluating a Resource Statement, Duplicate declaration: Service_wrapper[hostapd] is already declared in file /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/init.pp:270; cannot redeclare at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/interface_ap.pp:38  at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/interface_ap.pp:38:5  at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_network/manifests/process_interface.pp:62 on node maverick-raspberry.home

fnoop, 2017-08-14:
Aargh.  This was supposed to be a fix for this:
https://github.com/fnoop/maverick/issues/538

It's supposed to disable hostapd if it doesn't detect an AP interface definition and it worked fine during extensive testing, but obviously there's still a bug in it somewhere.  Thanks, wilil reopen :(

lucidwan, 2017-08-14:
It's all good. I'll do some digging in a bit.

Are you saying this function is for when it doesn't detect and AP? 
I have an AP interface defined and working on this one.

fnoop, 2017-08-14:
Yes, it's here:
https://github.com/fnoop/maverick/blob/stable/manifests/maverick-modules/maverick_network/manifests/init.pp#L269

The idea is that if it doesn't detect that AP class has been called (defined) then it stops hostapd, as otherwise it interferes with normal interface operation.  But puppet is a bit funny about detecting whether a class has been defined reliably, I think it's a non-deterministic ordering thing..

fnoop, 2017-08-14:
Could you try self-update and configure again?  I made a very small update that may or may not have any effect..

lucidwan, 2017-08-15:
Same error this time, just a line change in init.pp:268

fnoop, 2017-08-17:
OK this should be fixed now.  Could you please test?

lucidwan, 2017-08-19:
Ok sorry for the delay, had a crazy-busy week.

It appears to go through the majority of the configure - until the nginx portion in which it gives the errors in the attached image. This obviously has Grafana / MavCesium not working @localhost:6790 shows the following (I didn't make the links of course):

{{alert.title}}
Docs
Support Plans
Community
Grafana v4.3.2 (commit: ed4d170)
New version available!

Wifi Works, have eth0, wlan0 and wap0 working after the config - so that's fixed : )

[image]

fnoop, 2017-08-19:
@lucidwan Thanks for the update.  I really need to update the docs - I tend to do that once I've finalised a release - but grafana has moved behind a reverse proxy now.  If you just go to <hostname> in a browser (on the default port 80), you should see a link to Grafana.  And I had totally forgotten about mavcesium on port 6790! (But happily that should work again now).
For the nginx error, could you please attach the configure log file for the run?  You should find it in ~/var/log/maverick.

lucidwan, 2017-08-19:
Ok:

Was tired last night, but before I quit I  rebooted, re-ran the config after I changed some mavproxy settings - and it ran without error.
Port 80 and Dashboards working though. No Mavcesium.

Log is attached. I included several - in case.

Now I just have to sort out actually getting it to connect to my FC : ) Can't seem to get it working atm.
I need to power it from the copter instead I'm thinking. Probably a ground reference problem on the comms.

BTW - I appreciate the help and the work. Linux is far from my favorite thing to setup. I'm a hack and know enough to be dangerous..
I can..but I'm just not of fan of the whole -let's make the cli and commands the longest things ever- process. Having all of this available is a beautiful thing. I'm down to help if you need things checked (once I get it setup and communicating) Feel free to ask.

[LBMaverickLogs.zip](https://github.com/fnoop/maverick/files/1236489/LBMaverickLogs.zip)

lucidwan, 2017-08-20:
Got it connected. Finally tried all UDP and wound up 14573 connects.

fnoop, 2017-08-21:
OK you got the same nginx error I did, and like me it worked the second time.  Raised issue:
https://github.com/fnoop/maverick/issues/575

fnoop, 2017-08-21:
You're getting the same grafana wget executing every run as I sometimes get:
https://github.com/fnoop/maverick/issues/574

fnoop, 2017-08-22:
nginx error fixed, grafana wget is just an annoyance and being tracked in it's own issue.

Glad you like it so far, whole point of it is to make life easier for you :)

closed 2017-08-22 by fnoop