goodrobots/maverick #567: Install issues Odroid XU4

lucidwan, 2017-08-14:
Here are the issues:
Image writes fine boots fine.

As on the raspberry, don't use self-update first. In my case, I setup up wifi, then  wman0 and wap0, in localconf- ran maverick configure, and everything updated correctly.

If you update first
The first error is a set of boot issues, which also show up on the maverick configure update (self update or not):
the are (4) - unhandled faults - imprecise external abort (xxxxxx)
Had to use sudo -H with maverick configure, as it kept throwing out folder permission errors in .cache, and one other.
After self update the odroid install has several different issues.
after update - invalid class "java_home". Added this to tensorflow conf as an empty declaration which solved it.
Invalid class "nginx".  The modules folder has an empty folder for Nginx. 
This was simple enough to figure out once I figured out puppet and manifests, and the install path for modules using puppet. 
The issue with nginx is that it won't install the correct dependencies - from puppet, that I could find using puppet-nginx and jfryman-nginx. Jfryman throws a server::resource error. Installing puppet-nginx --force fixes it (adds the declaration to the .conf) 
So when you maverick configure, it throws out a ton of dependency errors.
I'm running fine - atm - without using self-update. I'll test this in a little bit.

I'm sure it's just that you're updating things which is awesome, but it's pulling down some broken code.

It's quite a little companion system you have..all the work is much appreciated.

fnoop, 2017-08-14:
I don't actually have an XU4 any more (decided life was too short to wait for odroid to support it properly), but would hope to keep support for it as long as possible as it's a popular platform.
The first error (imprecise external abort (xxxxxx)) is I think an odroid kernel warning, off the top of my head, I don't think that has anything particularly to do with maverick (other than providing the kernel), and I don't think it's fatal or there is anything you can do about it.

fnoop, 2017-08-14:
Re: the sudo -H issues, this should be fixed in the latest maverick code - if you look at /srv/maverick/software/maverick/bin/maverick, where it actually calls puppet at the end of the file it uses sudo -H now (main culprit is pip).  If you do a `maverick self-update` before you start it should run fine, and you shouldn't need to (and shouldn't) run maverick under sudo.
Of course this depends on me not breaking stable..

fnoop, 2017-08-14:
```
after update - invalid class "java_home". Added this to tensorflow conf as an empty declaration which solved it.
```
Yes this is to be expected - I threw my odroid out the window just before finishing tensorflow.  Tensorflow compile/install works on x86 and raspberry, but there are probably a few small things to fix on odroid.  Any PRs gratefully received :)

lucidwan, 2017-08-14:
Yeah I figured it was something but I was having trouble figuring out where it was being called : )
I'll mess around it with it. It's my first time using this Odroid. It gets super hot, and the fan doesn't stay running. Pretty sure it threw a fault on me because of it last night. I'll keep you informed. Thanks!

fnoop, 2017-08-17:
Could you please try with latest self-update, and see how it configures now?

fnoop, 2017-09-25:
Closing as I don't have an odroid any more.  If anyone wants to produce a downloadable image of a latest version, please do reopen this.

closed 2017-09-25 by fnoop