goodrobots/maverick #585: Configure endpoint IP's for mavlink-router.

vgroove, 2017-08-24:
According to the mavlink-router sample config UDP Endpoint addresses act as follows:

"If on `Normal` mode, IP to which mavlink-router will route messages to (and from)."

https://github.com/01org/mavlink-router/blob/master/examples/config.sample

Which allows users to enter a GCS IP address if you want to route to a different computer.  Currently in Maverick this value is just set to 127.0.0.1 and cannot be changed through the maverick configure command.  Any way to add in specific IP's or IP ranges besides manually editing mavlink-router-fc.conf everytime?

fnoop, 2017-08-24:
Good question.  Basically no, it's fixed in the config generation.  It could be made configurable potentially but they're left as 127.0.0.1 for on-board usage.  The idea with Maverick is to try and go back to a more normal/sane client/server setup where multiple clients connect to the central server (maverick).  To do this, mavlink-router is setup by default with 3 'normal' udp ports and 3 'eavesdropping' (called udpin by most other mavlink systems).  So instead of having to reconfigure the server every time a client changes IP address (which is insane), you just configure the client to connect to the server (if you're not using windows, you can use zeroconf dns like maverick-raspberry.local as the address instead of having to find out the IP address).  All GCSs these days should be able to connect to a specified udp address (look for udp client or udpci).  I've found this is a much, much easier way to do things.
But certainly open to others wishes/requirements, if it's something that would be actually useful then happy to look at implementing it.

fnoop, 2017-08-24:
List of ports here:
https://fnoop.github.io/maverick/#/modules/mavlink?id=proxy-ports
The above blurble could certainly be made clearer in the docs.

vgroove, 2017-08-24:
Alright, looks like I should rethink the way I've been connecting my GCS, that does sound easier!  Thanks for the help!

closed 2017-08-24 by vgroove