goodrobots/maverick #589: Multiple wifi network definitions

jkcoop17, 2017-09-13:
Hello,

Sorry if this is the wrong place to ask this, but I'm using maverick on a pi 3 and had a question.

I have multiple possible wifi networks I would like to connect to.  On a standard raspbian install I would just put all the network definitions in wpa_supplicant.conf, then have the wireless interface in /etc/network/interfaces point to wpa_supplicant.  This way it connects to whichever network is nearby, and if there are more than one you can set a priority.

Trying to get this working with maverick I have something like this in my localconf.json:

```
"wifi_ssid":    "network1",
"wifi_psk":    "blah",

"maverick_network::interfaces": {
                "wman0": {
                    "type":     "wireless",
                    "ssid":     "network1",
                    "psk":      "blah"
                },
                "wman1": {
                    "type":     "wireless",
                    "ssid":     "network2",
                    "psk":      "blahblah"
                },
                "wman2": {
                    "type":     "wireless",
                    "ssid":     "network3",
                    "psk":      "blahblahblah"
                },
                "wman3": {
                    "type":     "wireless",
                    "ssid":     "network4",
                    "psk":      "blahblahblahblah"
                }
},
```

If the network in "wifi_ssid" is in range it connects fine, but otherwise it won't use any other interfaces.  Is there a way to specify multiple possible networks?

I figured I could make a wpa_supplicant and then make a single wman0 definition in localconf.json that points to wpa_supplicant.  But it looks like that would get wiped after I call `maverick configure`, correct?

fnoop, 2017-09-13:
Very good question!  Yes you just want a single wman0 interface - you can't define multiple additional interfaces as they are physical interface definitions (you could possibly define interface aliases, but not advisable).  I'll have a think, possibly the easiest way round this is for maverick to leave wpa_supplicant.conf alone and let you alter it as you wish.  You can actually do this now by leaving "wifi_ssid" and "wifi_passphrase" blank/undefined - they both need to be defined otherwise maverick doesn't touch wpa_supplicant.conf (theoretically).  If you could try that and report back if it works, then I'll alter the manifests and documents to make it 'official' :)

jkcoop17, 2017-09-13:
I tried leaving wifi_ssid and wifi_passphrase both blank and just absent from localconf.json.  In both cases the interface definition still had something like:

```
allow-hotplug wlan0
iface wlan0 inet dhcp
    ssid
    psk
    ...other stuff...
```

my typical interface file has:

```
allow-hotplug wlan0
iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
So I tried adding a wpa-conf or wpa-roam options in localconf.json, but that gave an error when trying `maverick configure` saying it didn't have a valid mapping.

jkcoop17, 2017-09-13:
Just did a self-update to test out that change, I only have this in my localconf.json network-wise:

```
"maverick_network::interfaces": {
    "wman0": {
        "type":    "wireless"
    }
},
```

and end up with this in both my wlan0 and wman0 interfaces

```
auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
    wpa-ssid
    wpa-psk
    post-up /sbin/dhclient wlan0
    post-up /srv/maverick/software/maverick/bin/network-if-managed wlan0
```

which doesn't work in setting up the interfaces, but does if I just remove the wpa-ssid and wpa-psk.  So for some reason it's still adding that in there even though I see the <% if @wpa_ssid -%>
      wpa-ssid <%= @wpa_ssid %>
  <% end -%> in that erb file.  Am I missing something?

fnoop, 2017-09-13:
Yes I shouldn't have pushed that to stable branch before testing it, just killed my raspberry wifi.  Will go and try and find where in the house it's plugged in and reset it, then push a proper fix :)

jkcoop17, 2017-09-13:
Haha good luck!

fnoop, 2017-09-13:
@jkcoop17 Could you please do a self-update and configure, and see how it looks now?

jkcoop17, 2017-09-14:
Works perfect!  Thanks for your help!

closed 2017-09-15 by fnoop