goodrobots/maverick #590: libjpeg-turbo8-dev has no install candidate on Raspberry Pi

CountZ74, 2017-09-15:
[maverick.dev.2017-09-15-10-49.log.zip](https://github.com/fnoop/maverick/files/1306167/maverick.dev.2017-09-15-10-49.log.zip)

monkeypants, 2017-09-15:
In raspbian it's called libturbojpeg-dev

libturbojpeg1-dev in some versions, libturbojpeg0-dev on my rpi-zero running raspbian-lite. apt-getting libturbojpeg-dev should resolve to the correct one.

fnoop, 2017-09-15:
Thanks @CountZ74 and @monkeypants, these package names have always confused me..  Hope this commit fixes things.
@CountZ74, that log file is weird, I wouldn't expect it to be doing so many things on an update, including compiling gstreamer.  Anyway, could you to an 'apt update' before doing a self-update and configure again?  Most package problems are almost always because apt cache hasn't been updated.

fnoop, 2017-09-15:
Oh yes I remember now, using the package alias isn't idempotent:
```Notice: /Stage[main]/Maverick_mavlink/Package[libturbojpeg-dev]/ensure: created```

monkeypants, 2017-09-15:
Libturbojpeg and libjpegturbo are like some kind of emacs/xemacs hostile fork thing with sad faces. Don't go there. I'm sure Tridge chose the right one.

fnoop, 2017-09-15:
Oh goody.. So on Ubuntu, all I see is:
```
libjpeg-turbo-progs/xenial 1.4.2-0ubuntu3 amd64
libjpeg-turbo-test/xenial 1.4.2-0ubuntu3 amd64
  Program for benchmarking and testing libjpeg-turbo
libjpeg-turbo8/xenial,now 1.4.2-0ubuntu3 amd64 [installed]
libjpeg-turbo8-dbg/xenial 1.4.2-0ubuntu3 amd64
  Debugging symbols for the libjpeg-turbo library
libjpeg-turbo8-dev/xenial,now 1.4.2-0ubuntu3 amd64 [installed]
libturbojpeg/xenial,now 1.4.2-0ubuntu3 amd64 [installed]
```
There's no libturbojpeg-dev available

closed 2017-09-15 by fnoop