goodrobots/maverick #599: Maverick configure not taking effect on Pi Zero

jkcoop17, 2017-09-22:
Hello,

I'm having trouble getting maverick configure to take effect on a raspberry pi zero.  I have made changes to the localconf.json to disable most of the services, as I mostly just want to use mavlink-router and some custom code.  However, when running configure it hangs up on Grafana and skips all the other steps.

```
Notice: /Stage[main]/Maverick_analysis::Grafana/File[/srv/maverick/software/grafana]/ensure: created
Notice: /Stage[main]/Grafana::Service/Service[maverick-grafana]/ensure: ensure changed 'stopped' to 'running'
Notice: /Stage[main]/Maverick_analysis::Grafana/Exec[grafana-postdelay]: Triggered 'refresh' from 2 events
Error: /Stage[main]/Maverick_analysis::Grafana/Grafana_datasource[influxdb]: Could not evaluate: Connection refused - connect(2) for "localhost" port 6790
Notice: /Stage[main]/Maverick_analysis::Grafana/Grafana_dashboard[system_dashboard]: Dependency Grafana_datasource[influxdb] has failures: true
Warning: /Stage[main]/Maverick_analysis::Grafana/Grafana_dashboard[system_dashboard]: Skipping because of failed dependencies
```

I have tried running self-update, then removing the grafana dir from ~/software, then running configure but still see the problem.

fnoop, 2017-09-25:
Hmm, this is probably because the zero takes so long to do anything, the hacky 10 second delay after grafana startup is called is not enough.  This should only affect you if grafana is not installed/running before configure is run, so try starting grafana first (maverick start grafana), before running maverick configure.

fnoop, 2017-09-25:
If you just want to turn grafana off, set localconf parameter `"maverick_analysis::grafana": false`.

fnoop, 2017-09-25:
https://github.com/voxpupuli/puppet-grafana/issues/35

fnoop, 2017-09-25:
Note there has been recent work on a 'raspberry lite' install:
https://github.com/fnoop/maverick/blob/stable/conf/sample-nodes/maverick-raspberrylite.json
This next version 1.1.4 will have a downloadable raspberrylite image which will fit in 4gb storage and will have only very minimal services running.

closed 2017-09-25 by fnoop