goodrobots/maverick #600: rude shock when unable to login after fresh install

monkeypants, 2017-09-22:
after fresh install, reboot, first login; I was unable to login with my old user (pi@...). I can't `su` to them as mav either, did the password change (or perhaps did I just get my passwords mixed up)?

Am able to `sudo su - pi`, my stuff's still there. If this is intended behaviour, there is an expectation management bug in the bootstrap-maverick documentation page.

fnoop, 2017-09-22:
'rude shock' is a bit dramatic...

Yes, it resets the pi password because otherwise for fresh installs it complains about an insecure password and was causing functional problems.  Possibly it should check if the password is default first before changing it.

monkeypants, 2017-09-22:
I was a bit dramatic to emphasise the UX aspect. The point of the ticket is the surprise. If the doc said it was going to happen it wouldn't be a problem. "here, run this script that does a heap of magic - oh dear, now I can't log in".

Simplest good solution is a warning/explanation on the bootstrap-maveric doc.

FWIW it wasn't default password.

fnoop, 2017-09-23:
Yes definitely should be documented, and an existing password should be left alone unless it's the shipped default.

fnoop, 2017-09-23:
Lol btw the password for pi is 'wingman' like the mav account.  I'll add this to the docs and change behaviour so it leaves non-standard passwords alone.  Password management is still something that needs addressed (#408)

monkeypants, 2017-09-24:
> and change behaviour so it leaves non-standard passwords alone.

wow, yes. I didn't realise it was clobbering my diligently reset password every time. That's definitely annoying. If the password isn't non-standard, doesn't that mean it would be unchanged if you didn't leave it alone? I mean, why not just don't clobber passwords at all?

fnoop, 2017-09-25:
OK so this is just an understanding thing.  Maverick uses puppet to do most of the heavy lifting behind the scenes.  Puppet is a declarative configuration management system.  This basically means that it *takes over control and management of all resources that it knows about*.  This should probably be stated more explicitly and clearly in the documentation.  'Clobbering passwords' is considered a feature, a good thing, in this system.  The idea is that you are supposed to engineer a blueprint of the system, and all changes go into the blueprint, nothing is done on an individual system itself.  Then you run maverick/puppet and it applies those changes to an individual system.
In this case, you have to change the default pi password otherwise it whines noisily, and I seem to remember a functional impact.  So a bit of puppet code was added to take over control of the pi user to set this automatically, so you don't have to.  Unfortunately, passwords aren't yet handled well in maverick, in that there's no easy one-stop-place to manage them.  This is planned.

fnoop, 2017-09-25:
Pi user password controlled here:
https://github.com/fnoop/maverick/blob/stable/manifests/maverick-modules/maverick_hardware/manifests/raspberry.pp#L20

Original issue:
https://github.com/fnoop/maverick/issues/234

The options here are:
 - continue to manage the pi user password, which is settable anyway by localconf parameter "maverick_hardware::raspberry::pi_password"
 - don't manage the pi user password at all, remove the warnings (or keep the warnings)
 - manage the pi password, but only if it's still standard

fnoop, 2017-09-25:
For now, will keep management with a docs update: https://fnoop.github.io/maverick/#/modules/hardware?id=raspberry-pi

closed 2017-09-25 by fnoop