goodrobots/maverick #601: fail to install wxpython on rpi3 due to unmet build dependencies

monkeypants, 2017-09-24:
[this gyst](https://gist.github.com/monkeypants/1d806416f2b095d5fc57de847df3e8ba) is my output of failed `maverick configure` with this localconf.json on a rpi3/raspbian-lite:
```
{	
    "base::locale::timezone": "Australia/Sydney",
    "maverick_mavlink::cmavnode_install": false,
    "maverick_mavlink::mavlink_router_install": false,
    "maverick_mavlink::mavcesium_install": false,
    "maverick_mavlink::mavproxy_install": true,
    "maverick_mavlink::mavproxy_type": "source",
    "maverick_mavlink::mavproxy_source": "http://github.com/monkeypants/mavproxy.git",
    "classes": [
	"maverick_mavlink",
	"maverick_vision"
    ]
}
```
In attempt to debug this, I also tried to `pip install Gooey` from the CLI. It [failed thus](https://gist.github.com/monkeypants/a64e4e298e8baeaea67823b42dcc83c4) 

This looks like maybe a GL problem (headers not found), so I manually `sudo apt-get install freeglut3-dev` and tried again. This time pip started building wxpython and took ages to fail.

I note on the [Gooey README](https://github.com/chriskiehl/Gooey#quick-start) that they recommend

> Python 2 users must manually install WxPython! Unfortunately, this cannot be done as part of the pip installation and should be manually downloaded from the wxPython website.

I don't want to believe that (downloading is not the problem, it downloaded fine). The [how to build phoenix](https://github.com/wxWidgets/Phoenix#how-to-build-phoenix) page on the wxWidgets site gave the following list of dependencies, which I installed with apt-get:
```
dpkg-dev
build-essential
python2.7-dev # use appropriate Python version here
libjpeg-dev
libtiff-dev
libsdl1.2-dev
libgstreamer-plugins-base0.10-dev
libnotify-dev
freeglut3
freeglut3-dev
libsm-dev
libgtk2.0-dev
libwebkitgtk-dev

If you are building for GTK3 then you'll also need these packages and their dependencies:

libgtk-3-dev
libwebkitgtk-3.0-dev

If your Linux distribution has gstreamer 1.0 available then you can install the dev packages for that instead of the 0.10 version. For example:

libgstreamer-plugins-base1.0-dev
```  
(I added them all, including  the last 3. Probably didn't need to install the 0.10 gstreamer as well)

So then I tried again with `pip install Gooey` and this time, eventually ...
```
Collecting Gooey
  Using cached Gooey-0.9.3-py2.py3-none-any.whl
Collecting wxpython==4.0.0b1 (from Gooey)
  Using cached wxPython-4.0.0b1.tar.gz
Collecting six (from wxpython==4.0.0b1->Gooey)
  Using cached six-1.11.0-py2.py3-none-any.whl
Building wheels for collected packages: wxpython
  Running setup.py bdist_wheel for wxpython ... done
  Stored in directory: /srv/maverick/.cache/pip/wheels/85/83/96/515f900d1a4b7cdb7ad15af8a513959e981254bf196403618d
Successfully built wxpython
Installing collected packages: six, wxpython, Gooey
Successfully installed Gooey-0.9.3 six-1.11.0 wxpython-4.0.0b1
```
I lack the puppet-fu to declare those OS packages are required if you want cuav installed on a rpi, otherwise I would make a PR.

monkeypants, 2017-09-24:
I should mention:
 * `"maverick_mavlink::mavproxy_source": "http://github.com/monkeypants/mavproxy.git"` is currently identical to `http://github.com/monkeypants/ArduPilot/MAVProxy.git`, so this is not some weird artefact of my doing
 * now that `pip install Gooey` worked, `maverick configure` almost also ran to completion without errors. 

It got past wxpython and Gooey, and barfed on:
```
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: running build_ext
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: building 'cuav.camera.chameleon' extension
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: creating build/temp.linux-armv7l-2.7
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: creating build/temp.linux-armv7l-2.7/cuav
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: creating build/temp.linux-armv7l-2.7/cuav/camera
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-9xgeTe/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/lib/python2.7/dist-packages/numpy/core/include -Icuav/camera/include -I/usr/include/python2.7 -c cuav/camera/chameleon_py.c -o build/temp.linux-armv7l-2.7/cuav/camera/chameleon_py.o -std=gnu99 -O3 -mfpu=neon -O0
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: In file included from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1788:0,
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:                  from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:                  from /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:                  from cuav/camera/chameleon_py.c:3:
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: /usr/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:  #warning "Using deprecated NumPy API, disable it by " \
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:   ^~~~~~~
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: In file included from cuav/camera/chameleon_py.c:7:0:
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: cuav/camera/include/chameleon.h:30:31: fatal error: libusb-1.0/libusb.h: No such file or directory
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:  #include <libusb-1.0/libusb.h>
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns:                                ^
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: compilation terminated.
Notice: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
Error: '/usr/bin/python setup.py install' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_mavlink/Exec[compile-cuav]/returns: change from 'notrun' to ['0'] failed: '/usr/bin/python setup.py install' returned 1 instead of one of [0]
Notice: Applied catalog in 7679.47 seconds

-------------------------
WARNING: If this is the first bootstrap run, please reboot NOW to activate system changes.
If system is not rebooted after first bootstrap run, further runs may not work as expected.
After reboot, login as 'mav' user - default password is 'wingman'
-------------------------

Maverick finished, happy flying :)
```
that looks like it might be an easy one...

monkeypants, 2017-09-24:
after `sudo apt-get install libusb-1.0-0-dev`, then `maverick configure` ran to the end without throwing any errors.

monkeypants, 2017-09-24:
I suppose maybe somewhere near line 311 of ... I there should be something like this:
```
        } elsif $::operatingsystem == "Raspbian" {
            ensure_packages(["libdc1394-22-dev", "libturbojpeg-dev", "python-wxgtk3.0", "libusb-1.0-0-dev", 'dpkg-dev', 'build-essential', 'python2.7-dev', 'libjpeg-dev', 'libtiff-dev', 'libsdl1.2-dev', 'libnotify-dev', 'freeglut3', 'freeglut3-dev', 'libsm-dev', 'libgtk2.0-dev', 'libwebkitgtk-dev', 'libwebkitgtk-3.0-dev', 'libgstreamer-plugins-base1.0-dev'])
        }
```

fnoop, 2017-09-24:
I'm getting the same thing rebuilding an ubuntu VM, it must be something they've changed in gooey. (that's not something you hear yourself say every day)

monkeypants, 2017-09-24:
I think maybe [this commit](https://github.com/chriskiehl/Gooey/commit/4ad3d4e71cc783b9979d569e086a54db99bc48d5#diff-2eeaed663bd0d25b7e608891384b7298) from August 8 might be the culprit. It pinned Gooey to version `4.0.0b1` (Phoenix), I think that is not available as a wheel, so it needs to be compiled, so it has those deps.

fnoop, 2017-09-25:
Installing wxpython4 on raspberry kills it, it's trying to compile -j4.

fnoop, 2017-09-25:
Added facility to install exact pip version in #604, set gooey dependency to 0.9.2.3 by default which only needs wxpython3, which is installed as a dpkg and much faster/easier.  wxpython4 has a much larger dependency chain, which is needed by gooey 0.9.3.
maverick_mavlink::gooey_version localconf parameter can be changed to specify the gooey version.

fnoop, 2017-09-25:
Added code to set correct dependencies if gooey >= 0.9.3 specified, so it does install if specified.

closed 2017-09-25 by fnoop

monkeypants, 2017-09-26:
Re: above, it seems we tripped over this at the same time as @stephen-dade over in the cuav repo. His fix was to pin gooey rather than install deps to build it.