goodrobots/maverick #607: SITL still active in flight mode

fnoop, 2017-09-30:
SITL process is still active in flight mode.  SITL should never be active in flight mode.

fnoop, 2017-10-01:
Testing on ubuntuvm works fine switching between the environments:
@SamPes can you make sure you've done a self update, then do:
`maverick --env=dev configure`
`maverick --env=flight configure`
And see if the SITL is running?  If it is, can you attach the last couple of logs from ~/var/log/maverick?

SamPes, 2017-10-01:
Log is here:
[maverick.dev.2017-09-29-15-29.log](https://github.com/fnoop/maverick/files/1347140/maverick.dev.2017-09-29-15-29.log)

fnoop, 2017-10-01:
Looks like you didn't do an 'apt update' before running maverick.  Can you do an apt update, then try again?

SamPes, 2017-10-01:
Oops I think that might have been the wrong log. This is the latest after doing apt update:
[maverick.flight.2017-10-01-15-16.log](https://github.com/fnoop/maverick/files/1347199/maverick.flight.2017-10-01-15-16.log)

fnoop, 2017-10-12:
OK turns out this is correct behaviour, albeit confusing.  The reason SITL is still running in flight mode is because you have set "maverick_dev::sitl::sitl_active: true" in localconf.json.  Within the maverick config hierarchy, the localconf.json is top of the pile - it overrides everything else, including the config in the environments.  
SITL is turned off in the flight environment by setting "maverick_dev::sitl::sitl_active: false" in ~/software/maverick/conf/environments/flight/defaults.json, but this has a lower precedence than localconf.json.  If you comment out the "maverick_dev::sitl::sitl_active" in localconf.json, then the flight defaults.json takes over and SITL correctly disables.

fnoop, 2017-10-12:
Confirmed that turning off these localconf params reverts to expected behaviour.  Separate issue #609 raised for the very slow configure run on the zero, along with a circular action.

closed 2017-10-12 by fnoop