goodrobots/maverick #610: wifi doesn't work headless and no clear path to reset it to defaults

davidbuzz, 2017-10-16:
I use maverick-1.1.1-raspberry.img.xz on a headless raspberry pi, and used the standard headless RPi methodology for getting my Wifi working , which is to mount the SD image on a host PC and put a wpa_supplicant.conf in /boot. 
When I did this and then booted the RPI off the SDCard, it was unable to connect to the wifi headless, but I did see that the wpa_supplicant.conf had been moved off teh SD card, so I know the OS booted.

So I then attempted to diagnose the failure by attaching a screen and keyboard and running wifi-setup as instructed in the instructions, and this executed, but still did not connect to my wifi. 

I then looked around and found I had wlan0 that was connected but without any IP addigned, and that /etc/network/interfaces was also defining a wman0 that didn't exist, so I removed the "unneeded" parts from the "interfaces" file, and tried to run wifi-setup a second time.   

running wifi-setup a second time refused to run, and reported it was already configured.  with no obvious way to restart or try again, I was forced to reimag all 2GB of the SD card just because wifi wasn't working and there was no clear instructions on how to proceed from here. 

references:
https://fnoop.github.io/maverick/#/?id=installation
https://raspberrypi.stackexchange.com/questions/10251/prepare-sd-card-for-wifi-on-headless-pi

fnoop, 2017-10-16:
There have been quite a few fixes that will directly affect this behaviour since 1.1.1, eg:
https://github.com/fnoop/maverick/commit/6f7807ad4d5fd551ba0a2462fa548925e6fddf7d
https://github.com/fnoop/maverick/commit/0bfaea5785df8209e8edfc6a5dd3a03585851b65
https://github.com/fnoop/maverick/commit/1c179a81941e2596f7de8b0a09d4b0d90f03fb3c
https://github.com/fnoop/maverick/commit/4c9dc66fa510fbf8be7ea333d6ba1d50b6d75062

If you do an update, which should be the first thing you do in Maverick (needs adding to the docs if not already there), what you're trying to do should work:
```
maverick self-update
maverick configure
```
Version 1.1.4 which has a lot of raspberry updates is due to go out sometime this week - you can try the master branch if you're feeling adventurous:
https://fnoop.github.io/maverick/#/?id=maverick-code-branch

But the latest stable should a) run wifi-setup even after modifying settings, and b) should honour wpa_supplicant.conf and not overwrite it.

The wlan0 interface is defined as a 'fallback' dhcp interface, as is eth0.  wman0 is intended to be the primary defined interface and needs the macaddress correctly updated in ~/data/config/maverick/localconf.json before doing a configure run.  wifi-setup does attempt to set this, and is sometimes successful. You usually need to reboot after running wifi-setup.  If you get wlan0 running but no IP address, sometimes running 'sudo dhclient wlan0' will get your IP address, again there is a fix for this in more recent versions of Maverick.

Please try again with the updates and see how you go.  I normally set Raspberrys up by plugging ethernet in, I'll try it SD card/wifi only as well.

davidbuzz, 2017-10-16:
thx for the feedback, I guess it's time you update the online instalation guide so it doesn't point to 1.1.1 then, as I downloaded it today.  :-)   I have now done a 'maverick self-update'  and  ' configure' as instructed.   I haven't checked ( it's still running configure etc), but I'm going to speculate that this is fixed in 1.1.4 and that the documentation download link is all that needs fixing now.  :-)

closed 2017-10-16 by davidbuzz

fnoop, 2017-10-16:
There is no later released version :)  The OS image downloads are intended just to get people going and aren't released every version, I should add a clear note to do a self-update before doing anything else.  I haven't udpated the docs for ages, I'll do a thorough review at 1.1.4 release.  Thanks for the feedback :)