goodrobots/maverick #611: Break maverick-analysis out into separate project

fnoop, 2017-10-16:
Separate maverick-analysis, or at least the graphing stuff, into a separate project so it can be used outside of maverick.
Alternatively test a node config that only turns on the analyis.

SamuelDudley, 2017-10-16:
Sounds great! We will leverage this to support free public log analysis for those that don't have maverick installed.

fnoop, 2017-10-16:
Need a new non-destructive environment for desktop/cloud deployments - #615

davidbuzz, 2017-10-16:
Perhaps you could have all changes to the filesystem that are outside of the homefolder ( /srv/maverick ) to be disabled by some sort of boolean configuration flag?

fnoop, 2017-10-16:
Yes, that's a good idea

fnoop, 2017-10-16:
So as a temporary effort which will be good enough for testing (it produces a very similar outcome to a fully engineered solution), I've added a 'maverick-analysis' node config which enables only the analysis module - influxdb, grafana and collectd.  To install this into any ubuntu/debian server, do something like:

Stage 1 (as a normal user):
```
sudo apt -y update
sudo apt -y install git
cd /var/tmp/
git clone https://github.com/fnoop/maverick.git
sudo maverick/bin/maverick --env=bootstrap configure
```

Stage 2 (login as mav user):
```
echo 'MAVERICK_BRANCH=master' >~/data/config/maverick/maverick-branch.conf
maverick self-update
maverick configure
maverick status
```

Note, this will still make some amendments to the core system as part of the mandatory base class.  You can see the types of updates from the attached initial bootstrap log.
[Uploading maverick.bootstrap.2017-10-16-11-21.log](...)

You might want to take a backup of /etc/hosts, and it will fiddle with your login motd, but apart from that shouldn't do anything to intrusive to the system.  Note since the above bootstrap log I've disabled by default the management of the root password, so it will leave that alone.

SamuelDudley, 2017-10-17:
Great! I'll spin up a new server on digital ocean and see how it goes. Is there a recommended version of Ubuntu to use to minimise pain? 16.04 or more recent?

fnoop, 2017-10-17:
Yes 16.04 ideally, I haven't tested 17.04 yet - #613

fnoop, 2017-11-23:
Added a new 'minimal' environment in #615 that doesn't touch anything in the system other than the bare minimum to create and run a minimal maverick environment in /srv/maverick.  This should be safe to run on any existing environment, as everything is self contained within the mav user and /srv/maverick directory.

fnoop, 2017-11-23:
Steps to bootstrap maverick analysis only, without touching the rest of an installed system and without the rest of maverick:
```
sudo apt update && apt -y install git
git clone https://github.com/fnoop/maverick.git --depth 1
sudo maverick/bin/maverick --env=minimal configure
```
Log out, log back in as 'mav' user
```
cp ~/software/maverick/conf/sample-nodes/maverick-analysis.json ~/config/maverick/local-nodes/`hostname`.json
maverick configure
```

These steps are for when current master (1.1.5-dev) is released to stable.  Current steps while in master are:
```
sudo apt update && apt -y install git
git clone https://github.com/fnoop/maverick.git -depth 1 -b master
sudo maverick/bin/maverick --env=minimal configure
```
Log out, log back in as 'mav' user
```
cd ~/software/maverick
maverick unshallow
git fetch
git checkout master
cp ~/software/maverick/conf/sample-nodes/maverick-analysis.json ~/config/maverick/local-nodes/`hostname`.json
maverick configure
```

fnoop, 2017-11-27:
Separate analysis module largely complete. Description field added, GPS anonymisation added, simple web logfile uploader added to temporary maverick website and to public demo website:
http://www.maverick.one/

closed 2017-11-27 by fnoop

DavidIngraham, 2018-02-12:
HI @fnoop, are the instructions above current for installing maverick-analysis outside of the maverick environment?

fnoop, 2018-02-13:
They should be, if you have any problems please raise them here and I'll fix them!