goodrobots/maverick #614: during 'maverick configure ' , nginx fails to be depoyed/configured.

davidbuzz, 2017-10-16:
( I'm guessing an 'apt-get update'  will fix it, but don't know yet. ) 

$maverick configure 

Maverick - UAV Companion Computer System - Version 1.1.3

Environment marker set and is being used to set maverick environment: dev
Maverick Environment:    dev 
Proceeding to update system configuration - please be patient, this can take a while..

Notice: Compiled catalog for maverick-raspberry.home in environment dev in 24.21 seconds
Notice: /Stage[bootstrap]/Base::Users/User[root]/password: changed password
Error: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install nginx' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  dmeventd libapparmor1 libdbi1 libdevmapper-event1.02.1 libelf1 libesmtp6
  liblua5.1-0 liblvm2app2.2 liblvm2cmd2.02 libmemcached11 libmodbus5 liboping0
  libow-2.9-8 libowcapi-2.9-8 libprotobuf-c1 librabbitmq1 libreadline5 librrd4
  libtokyocabinet9 libtokyotyrant3 libtxc-dxtn-s2tc0 libupsclient4 libvirt0
  libyajl2 lvm2 owfs-common rrdtool
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  nginx-common nginx-full
Suggested packages:
  fcgiwrap nginx-doc ssl-cert
The following NEW packages will be installed:
  nginx nginx-common nginx-full
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 535 kB of archives.
After this operation, 1226 kB of additional disk space will be used.
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx-common all 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx-full armhf 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx all 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx-common_1.6.2-5+deb8u4_all.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx-full_1.6.2-5+deb8u4_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx_1.6.2-5+deb8u4_all.deb  404  Not Found [IP: 93.93.128.193 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Error: /Stage[main]/Nginx::Package::Debian/Package[nginx]/ensure: change from 'purged' to 'present' failed: Execution of '/usr/bin/apt-get -q -y -o DPkg::Options::=--force-confold install nginx' returned 100: Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  dmeventd libapparmor1 libdbi1 libdevmapper-event1.02.1 libelf1 libesmtp6
  liblua5.1-0 liblvm2app2.2 liblvm2cmd2.02 libmemcached11 libmodbus5 liboping0
  libow-2.9-8 libowcapi-2.9-8 libprotobuf-c1 librabbitmq1 libreadline5 librrd4
  libtokyocabinet9 libtokyotyrant3 libtxc-dxtn-s2tc0 libupsclient4 libvirt0
  libyajl2 lvm2 owfs-common rrdtool
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  nginx-common nginx-full
Suggested packages:
  fcgiwrap nginx-doc ssl-cert
The following NEW packages will be installed:
  nginx nginx-common nginx-full
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 535 kB of archives.
After this operation, 1226 kB of additional disk space will be used.
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx-common all 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx-full armhf 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx all 1.6.2-5+deb8u4
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx-common_1.6.2-5+deb8u4_all.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx-full_1.6.2-5+deb8u4_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/nginx/nginx_1.6.2-5+deb8u4_all.deb  404  Not Found [IP: 93.93.128.193 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Notice: /Stage[main]/Nginx::Config/File[/etc/nginx]: Dependency Package[nginx] has failures: true
Warning: /Stage[main]/Nginx::Config/File[/etc/nginx]: Skipping because of failed dependencies

fnoop, 2017-10-16:
If you could confirm that an apt update fixes it, that would be useful.  In general (and the docs do note this), you have to make sure the apt db is up to date before running maverick.  Apt isn't very good at keeping itself in a good state as it blindly relies on it's cache.  For custom/added apt repos maverick knows to trigger an apt update before installing, but for system packages (like nginx) it assumes the packages are available.

fnoop, 2017-10-16:
If there was a quick/easy way to check apt is up to date at the start of a configure run I'd be very glad to add it, but when I looked briefly before I couldn't see anything obvious.

fnoop, 2017-10-16:
Note: the install docs need updating to add an 'apt update' step before the initial self-update and configure.

davidbuzz, 2017-10-16:
I can confirm that 'apt update' resolved that specific problem.  (  other problems persist and I still don't have the system running tho  ).

fnoop, 2017-10-16:
Added note to do 'apt update' before initial configure run, also updated maverick command to check if update is necessary, #621

closed 2017-10-16 by fnoop