goodrobots/maverick #621: Can ‘apt update’ be detected if update is necessary?

fnoop, 2017-10-16:
Is it possible to detect if an update is necessary, without doing an update?

fnoop, 2017-10-16:
'when logging in to Ubuntu the login message normally shows if there are packages that need updating'
Determine mechanism

novirium, 2017-10-16:
Looks like there's a python script at `/usr/lib/update-notifier/apt-check`.
https://askubuntu.com/a/457887

Haven't looked into the exact mechanism, but the output of that may well be sufficient for this use

fnoop, 2017-10-16:
/usr/lib/update-notifier/apt-check is interesting but works off cache and not available by default in raspbian/debian

fnoop, 2017-10-16:
Whenever apt update is run, it updates /var/lib/apt/periodic/update-success-stamp.  This can be used to tell if an update has been done in the past day or two as one of the maverick checks before configure is run, and if not then to call an update.  Additionally, apt should be set to periodically update the cache lists frequently.
Neither of these, even together, will completely prevent package errors during a run, but it should significantly reduce them.

fnoop, 2017-10-16:
Update-Package-Lists set to 1 day update in /etc/apt/apt.conf.d/10periodic so should update daily.  maverick will now update if not.

fnoop, 2017-10-24:
/var/lib/apt/periodic/update-success-stamp isn't created on raspberry, so apt update is called every run.

fnoop, 2017-10-24:
Raspbian doesn't have /etc/apt/apt.conf.d/15update-stamp so it doesn't update /var/lib/apt/periodic/update-success-stamp.

closed 2017-10-24 by fnoop