goodrobots/maverick #624: Feature Request - Make Maverick manual available locally via webpage

SamPes, 2017-10-16:
Make Maverick manual documents available via a webpage via Nginx.
Optionally add developer or user based documentation for services or checklists, possibly via a webpage based text editor.

fnoop, 2017-10-20:
docsify supports service workers but not selfcerts :( Have to find a way round this as it will be crucial for web gcs. Getting sw to work on maverick-docs is an important proof of concept.

fnoop, 2017-10-20:
Possible options are:
 - Adding another secure origin
 - Adding cert chain to browser with documentation
 - Starting browser with insecure option or changing flag

davidbuzz, 2017-10-20:
why does docsify even need to know about ssl or certificates..?   Nginx can be the front-end that does the certs and user interface, and reverse-proxy in-front of a non-secure ( http) docsify instance that only listens on a locahost port etc. ..? or did I misunderstand..?

fnoop, 2017-10-20:
@davidbuzz for service workers.  docsify itself doesn't care where it's running from, but to support the service worker for complete offline use you need to have a trusted secure origin or a trusted certificate chain - a self signed cert doesn't work.  As service workers will be crucial to the web gcs, it's really important to get this working here first.
There's no docsify instance running on maverick, it serves the files directly.

fnoop, 2017-10-20:
OK I think importing the cert into the browser/OS is probably the best way to go, eg:
https://github.com/lwsjs/local-web-server/wiki/How-to-get-the-%22green-padlock%22-with-a-new-self-signed-certificate
It's the most 'correct' way and doesn't involve screwing around with browser security or startup flags

fnoop, 2017-10-20:
This should work but doesn't:
https://gist.github.com/WebReflection/b4b460ecfc92ee62a9a8
https://www.webreflection.co.uk/blog/2015/08/08/bringing-ssl-to-your-private-network

Need to figure out how to generate a self signed cert that can be trusted by the browser/OS, and a compatible nginx configuration.

fnoop, 2017-10-21:
Recent browsers - chrome/firefox have deprecated common name matching in certs, must use SAN now.  Going to add a local CA in #628 anyway, as it will be useful going forward.
Closing this as have verified that with a correct certificate setup, maverick docs service worker is working as expected.

fnoop, 2017-10-22:
Offline access working for accessed pages, need to precache all pages

fnoop, 2017-10-23:
service worker needs rewritten, will do later with web gcs framework.  For now doc pages that have not been accessed are not cached and won't be available offline.

closed 2017-10-23 by fnoop