goodrobots/maverick #638: Take a closer look at resionOS

fobdy, 2017-10-26:
https://resinos.io
In theory this could be a good base platform for maverick goals.

auturgy, 2017-11-14:
I disagree. Yocto is just painful.

fnoop, 2017-11-14:
@fobdy thanks for the input :)  At first glance resinos does look very interesting and potentially a good base for a system like maverick.
However, as @auturgy points out yocto is painful - not just painful I would rather stick needles in my eyeballs than use yocto.  I would rather use a system designed by my cat than yocto.  Yocto is really designed for hardware/system vendors to create fixed embedded systems that go into commercial products like cars or TVs or .. anything that users won't directly interact with.  It's really, truly horrible for developers or users to use.
Then they've combined this with docker containers, which were the next big thing until people realised it's limitations and drawbacks.  Containers are a great technology but are inherently unnecessary for an embedded/onboard system.  Docker came out of devops and has it's own variant of orchestration but isn't nearly as good as the likes of puppet/chef/ansible when it comes to configuration management of existing systems.
Which is the main point here - most embedded systems have their own OS variants which are customised to optimally support the hardware.  Raspberry, Odroid, Up, Tegra etc - they all have their own OS variants that are extensively customised to support the hardware platform.  Abstracting this out to a system like resinos would be great if they all committed to it, but they haven't, so resinos tries to fit all systems and will do so to a limited degree because of that.
Instead, Maverick leverages each platform's customised OS and then builds a common platform on top of them.
Will keep an eye on how the likes of resinos and coreos (although that's really aimed at bigger platforms), thanks for the suggestion.

closed 2017-11-14 by fnoop