goodrobots/maverick #647: Add python3 support throughout

fnoop, 2017-11-01:
Python 3 modules should be installed alongside python 2 modules

fnoop, 2017-12-03:
Break python out from base::packages into separate base::python class.

PYTHONPATH is set in various places in the manifests to scripts in /etc/profile.d, all to python 2.7 directories.

Components that need python 3 support:
 - ROS
 - Gstreamer
 - OpenCV

fnoop, 2017-12-04:
Adding python3-rospkg creates a bounce with ros packages.

```
Notice: Compiled catalog for maverick-up.home in environment dev in 13.59 seconds
Notice: /Stage[main]/Maverick_ros/Package[python3-rospkg]/ensure: created
Notice: Applied catalog in 48.29 seconds
```

```
Notice: Scope(Class[Maverick_ros]): ROS: supported platform detected for lunar distribution, using native packages
Notice: Compiled catalog for maverick-up.home in environment dev in 13.64 seconds
Notice: /Stage[main]/Maverick_ros/Package[python-rosdep]/ensure: created
Notice: /Stage[main]/Maverick_ros/Package[ros-lunar-perception]/ensure: created
Notice: /Stage[main]/Maverick_ros/Package[ros-lunar-viz]/ensure: created
Notice: /Stage[main]/Maverick_ros/Package[ros-lunar-mavros]/ensure: created
Notice: /Stage[main]/Maverick_ros/Package[ros-lunar-mavros-extras]/ensure: created
Notice: /Stage[main]/Maverick_ros/Package[ros-lunar-test-mavros]/ensure: created
Notice: Applied catalog in 89.00 seconds
```

fnoop, 2018-02-02:
ROS needs python3-rospkg-modules python3-catkin-pkg-modules instead pf python3-rospkg

fnoop, 2018-02-02:
install_python_module puppet helper defined in ~/software/maverick/manifests/init.pp uses python::pip and only installs python2.  Need to either adapt to install python3 as well, or else as an option.

SamuelDudley, 2018-09-17:
The goal should be to support Python 3.5+where python3 is used. Older versions of python3 are not forward compatible.

fnoop, 2018-09-17:
Need to set ROS_PYTHON_VERSION to 3 to build with python3.  Python 3.5 is apparently a minimum version, and not all packages are supported.  With ROS Melodic more transition is made to python 3 and almost all core and major packages should work with both python 2 and 3.
So in summary with ROS1 it looks like Python 2 is the current supported version and will remain until at least the next major version of ROS (after Melodic), but some stuff should work with Python 3.
Need to check if MAVROS works with python 3 yet with Melodic, last time I checked it didn't work at all:
https://github.com/mavlink/mavros/issues/940

SamuelDudley, 2018-09-18:
Pymavlink works with python3 but supporting tools such as dflogger are not yet working.

fnoop, 2018-09-18:
I don't think any of our projects use dflogger, or pymavlink.  don't think they should be dependencies.

SamuelDudley, 2018-09-18:
I'm still keen to have a pymavlink API option exposed for use, but making the ROS option cleaner and more user friendly is my primary concern.

fnoop, 2018-09-18:
Can we always have pymavlink as an optional extra, rather than something core?  As I understand it pymavlink is responsible for mavproxy and dronekit hammering cpu on low power boards such as the raspberry.  It's one of the main drivers towards ROS which is a lot less resource intensive.

fnoop, 2018-11-28:
Need to add python3-pip to python manifest

fnoop, 2018-11-28:
and python3-future - surprised this isn't installed by default with python3

SamuelDudley, 2018-11-28:
The future module shouldn't be needed for 3.6+ code as most (if not all) of the functionality of the future module is in the core library.

fnoop, 2018-11-28:
It's to provide backwards compatibility with python2 for python3 fixes, for example mavros/param.py fix for iteritems() in python is to use:
```
from future.utils import viewitems
```
and then
```
viewitems(params)
```

fnoop, 2019-03-27:
Completed in #809

closed 2019-03-27 by fnoop