goodrobots/maverick #660: Add zerotier support

fnoop, 2017-11-10:


SamuelDudley, 2017-11-17:
If running ZeroTierOne It would be nice to provide an interface to the JSON API which is exposed on 127.0.0.1:9993 for configuration and control. API details here: https://github.com/zerotier/ZeroTierOne/tree/master/controller

fnoop, 2017-11-20:
@SamPes @SamuelDudley should zerotier default to the public Earth network, or not join anything until network id specified?

SamuelDudley, 2017-11-20:
I'm voting for joining the Earth network. Instant working features are great for people experimenting with the software. Even if it's just a test to know everything is working as expected... Maybe opt in for VPN support?

SamPes, 2017-11-21:
Joining Earth network is fine provided it's an option that can be easily
disabled in the forthcoming webgui. ;-)

The alternative is that a instructions are written to setup your own ZT
network on Maverick and you can enable it with a line in Maverick config.
(Plus maybe a few lines to specify which Network ID you want to join.) To
complete the config the user still has to use ZT Central to setup the ZT
network and enable the clients. Which is likely required anyway and is how
most will operate it unless we setup our own ZT servers for Maverick.
(which is only viable to do in the future but not now)

From what I understand the Earth network is "public" so other's on the same
earth network can ping it, which is a security issue, which also makes it
easy to connect to....by people you don't want! :-)
The point being that it sets a insecure default precedent many users will
ignore and leave the front door open to ZT earth. We'd recommend against
using a default connection to ZT Earth atm.

From our testing so far there's a couple things we noticed.
1) ZT is particularly useful for NATed connections like 4G, or changing
public IPs where it offers the easiest way to connect.
2) When running ZT has a encryption/CPU overhead - this can have
detrimental impacts on performance if for example streaming high data rate
video.
3) When running on a local network like wifi or LAN ZT is superfluous -
wifi is still more common that running 4G atm

All of the above factors also would benefit from ZT not being enabled by
default at all, let alone running on the earth network.

For a stand alone GCS application or app etc this could be used to
incorporate ZT:  https://github.com/zerotier/libzt

SamD's idea of adding support to Maverick users by allowing them to connect
to a Maverick ZT network for devs to connect to for support reasons would
be useful. The easiest way for that atm would be that Fnoop could manage
his own ZT networks from ZT Central and get users to add that ZT network ID
to their Maverick systems to allow Fnoop to SSH in etc. This would also
become a part of the webgui config etc. But eventually a "support network"
system would be good to implement that only requires users to authorise
access.

Overall it's probably good to have a more complete conversation about how
to deal with comms in general?

Maybe a good time to open a Comms channel on gitter to discuss?

fnoop, 2017-11-21:
So the options are:
 - Join the earth network, but don't start the service by default
 - Join the earth network, start the service by default and accept security risk
 - Create a 'maverick' network and join everyone to it by default
 - Don't create/join any network, just install the software and leave it up to the user
---

> - Create a 'maverick' network and join everyone to it by default

I think this is a terrible idea for two reasons:
 1) It creates a 'backdoor', eroding trust in the system
 2) Creating a 'support' network/entry marks Maverick as a supported/commercial product.  It's absolutely not, and never will be (from me, anyway).  It's an opensource tool for others to use as they see fit.
 3) It creates an automated paradise for hackers - join the maverick network and there's a ready set of installs probably with lots of default logins available
 4) Provides no benefit over the earth network
OK that's four reasons!

> - Don't create/join any network, just install the software and leave it up to the user

This goes against the spirit of Maverick a bit, which is to provide a turnkey, out of the box experience.  Although it's certainly a valid option.

> - Join the earth network, start the service by default and accept security risk

I don't know if you can scan earth nodes, but if you can then joining default logins to it is a really bad idea.  If not, then this would provide a nice OOB experience.

> - Join the earth network, but don't start the service by default

This would probably be the easiest/safest compromise?

fnoop, 2017-11-21:
@SamuelDudley As I understand it the json api is for managing the controllers, not the client/service itself?  Running a controller is an interesting idea, but probably out of scope of this initial implementation.

SamuelDudley, 2017-11-21:
Ah I think I should have been more clear with my initial thoughts. By opt in VPN support I meant that users would need to opt in to engage the VPN functionality of Maverick (rather than user support via VPN). 
As SamP stated not all interfaces will require the VPN option so opting in means that the user is already doing an action before the default connection is attempted. Perhaps at that point having a default Earth connection is moot as the whole experience is no longer automagical?

SamuelDudley, 2017-11-21:
@fnoop Sorry, you posted while I was typing! I'm in favour of shipping with the Earth network config but not enabling the service as an initial configuration. The default service state can always be reviewed to on at a later date if Earth is considered safe enough.
Looking at the API again, yes you are correct. I think they are using this API to enable the ZT Central functionality. As you stated, useful if running your own server but out of scope.

fnoop, 2017-11-21:
@SamuelDudley That's cool!  Yes definitely useful for webui, or at least the backend of it.

fnoop, 2017-11-21:
OK added initial support for zerotier.  It installs the zerotier gpg key, repo and package, creates the crypto keys, and enables non-sudo use from mav user.

There are four options:
 - maverick_network::zerotier::libzt - defaults to false, if set to true will download and compile libzt
 - maverick_network::zerotier::active - defaults to false, if set to true will run the service and set to run at boot
 - maverick_network::zerotier::join_network - defaults to true, if set to true will join the network specified in $maverick_network::zerotier::network_id.  If set to false will leave the network
 - maverick_network::zerotier::network_id: defaults to the public earth network.  Change to a private ID to join a private network.

closed 2017-11-21 by fnoop

SamuelDudley, 2017-11-21:
Nice! :+1: