goodrobots/maverick #662: Raspberry pi won’t boot off USB

fnoop, 2017-11-12:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

lvale, 2017-11-12:
don't waste too much time on this. Although it's a nice to have while testing, I haven't debug this enough.

lvale, 2017-11-12:
and the milestone, should be "in a galaxy, far far away..."

fnoop, 2017-11-12:
It would actually be pretty cool to have working!  It takes forever to reflash sd cards while testing..

fnoop, 2017-11-17:
Confirmed,  Odd, it must be something do with the way the maverick images are recreated, as they are based off the same raspbian image that does work.
This will be great for building and testing maverick so should be fixed.

fnoop, 2017-11-21:
Look at how pi-gen does it
https://github.com/RPi-Distro/pi-gen

fnoop, 2018-03-27:
raspberry fact error:
```
Mar 27 18:32:14 maverick-raspberry systemd[1]: Starting Maverick MOTD Creator...
Mar 27 18:32:18 maverick-raspberry sh[1641]: Traceback (most recent call last):
Mar 27 18:32:18 maverick-raspberry sh[1641]:   File "/srv/maverick/software/maverick/bin/maverick-info", line 72, in <module>
Mar 27 18:32:18 maverick-raspberry sh[1641]:     print twocols('SD Card Size', '\t\t' +str(raspberry.data['sdsize']) +' Mb')
Mar 27 18:32:18 maverick-raspberry sh[1641]: KeyError: 'sdsize'
Mar 27 18:32:18 maverick-raspberry systemd[1]: maverick-motd.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 18:32:18 maverick-raspberry systemd[1]: Failed to start Maverick MOTD Creator.
Mar 27 18:32:18 maverick-raspberry systemd[1]: maverick-motd.service: Unit entered failed state.
Mar 27 18:32:18 maverick-raspberry systemd[1]: maverick-motd.service: Failed with result 'exit-code'.
```

fnoop, 2018-03-30:
1.1.6 release flashes and runs fine from USB now

closed 2018-03-30 by fnoop