goodrobots/maverick #671: Add support for up boards/ubilinux

fnoop, 2017-11-16:


fnoop, 2017-11-17:
```
Notice: /Stage[main]/Maverick_analysis::Influx/Exec[influx-repo]/returns: /etc/os-release: line 7: 34: command not found
Error: '/bin/bash -c 'source /etc/os-release; test $VERSION_ID = "7" && echo "deb https://repos.influxdata.com/debian wheezy stable" | sudo tee /etc/apt/sources.list.d/influxdb.list; test $VERSION_ID = "8" && echo "deb https://repos.influxdata.com/debian jessie stable" | sudo tee /etc/apt/sources.list.d/influxdb.list; test $VERSION_ID = "9" && echo "deb https://repos.influxdata.com/debian stretch stable" | sudo tee /etc/apt/sources.list.d/influxdb.list'' returned 1 instead of one of [0]
Error: /Stage[main]/Maverick_analysis::Influx/Exec[influx-repo]/returns: change from 'notrun' to ['0'] failed: '/bin/bash -c 'source /etc/os-release; test $VERSION_ID = "7" && echo "deb https://repos.influxdata.com/debian wheezy stable" | sudo tee /etc/apt/sources.list.d/influxdb.list; test $VERSION_ID = "8" && echo "deb https://repos.influxdata.com/debian jessie stable" | sudo tee /etc/apt/sources.list.d/influxdb.list; test $VERSION_ID = "9" && echo "deb https://repos.influxdata.com/debian stretch stable" | sudo tee /etc/apt/sources.list.d/influxdb.list'' returned 1 instead of one of [0]
Notice: /Stage[main]/Maverick_analysis::Influx/Exec[influx-aptupdate]: Dependency Exec[influx-repo] has failures: true
Warning: /Stage[main]/Maverick_analysis::Influx/Exec[influx-aptupdate]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_analysis::Influx/Package[influxdb]: Dependency Exec[influx-repo] has failures: true
Warning: /Stage[main]/Maverick_analysis::Influx/Package[influxdb]: Skipping because of failed dependencies
Notice: /Stage[main]/Maverick_analysis::Influx/File[/srv/maverick/data/config/analysis/influxdb.conf]: Dependency Exec[influx-repo] has failures: true
```

fnoop, 2017-11-17:
```
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: Err:9 http://packages.ros.org/ros/ubuntu dolcetto/main amd64 Packages
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns:   404  Not Found [IP: 140.211.166.134 80]
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: Ign:10 http://packages.ros.org/ros/ubuntu dolcetto/main Translation-en
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: Ign:12 http://packages.ros.org/ros/ubuntu dolcetto/main Translation-en_GB
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: Reading package lists...
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: W: The repository 'http://packages.ros.org/ros/ubuntu dolcetto Release' does not have a Release file.
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: E: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/dolcetto/main/binary-amd64/Packages  404  Not Found [IP: 140.211.166.134 80]
Notice: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: E: Some index files failed to download. They have been ignored, or old ones used instead.
Error: '/usr/bin/apt-get update' returned 100 instead of one of [0]
Error: /Stage[main]/Maverick_ros/Exec[ros-aptupdate]/returns: change from 'notrun' to ['0'] failed: '/usr/bin/apt-get update' returned 100 instead of one of [0]
```

fnoop, 2017-11-17:
Opencv build fails:
```
[ 34%] Building CXX object modules/flann/CMakeFiles/opencv_test_flann.dir/test/test_lshtable_badarg.cpp.o
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
modules/core/CMakeFiles/opencv_perf_core.dir/build.make:86: recipe for target 'modules/core/CMakeFiles/opencv_perf_core.dir/perf/opencl/perf_arithm.cpp.o' failed
modules/flann/CMakeFiles/opencv_test_flann.dir/build.make:62: recipe for target 'modules/flann/CMakeFiles/opencv_test_flann.dir/test/test_lshtable_badarg.cpp.o' failed
```

fnoop, 2017-11-17:
dhclient on ublinux not writing /etc/resolv.conf, so no dns resolution

fnoop, 2017-11-17:
/etc/resolv.conf a leftover symlink to connman:
```lrwxrwxrwx 1 root root 24 Nov 17 14:10 /etc/resolv.conf -> /run/connman/resolv.conf```

fnoop, 2017-11-17:
A lot of memory errors while compiling.  No swap, so add a gb swap file

fnoop, 2017-11-19:
Up core has no ethernet port so difficult to setup wifi without screen.  Would be nice to have similar /boot/wpa_supplicant mechanism that raspberry has.

fnoop, 2017-11-19:
Librealsense fails during bootstrap run

fnoop, 2017-11-19:
x11 auth window pops up during wifi-setup

fnoop, 2017-11-19:
Clean bootstrap and dev configure run now, but cloud9 and mavros services failing.

cglusky, 2017-11-19:
on Up^2 using 1.1.5 beta image.  

Windows 10 etcher write to 8GB USB stick(~10 minutes with ~5 minute verify)
Plug into up^2
Power on and select defaults for clonezilla
Wrote to mmc
Reboot/remove USB
logged in
status showed normal default services running (maverick is identifying Aeon Up^2 in status as well)
wifi setup (running an intel 7260 on the mini pcie bus)
reboot and connected to network

works fine

tested cloud9 via browser  works
tested grafana via browser not seeing anything in system display

ran top on local cli and seeing "arducopter" using 95%+ cpu and maverick-vision using about 50%

tested r200 with visiond and it works using video2 for streaming

fnoop, 2017-11-19:
@cglusky Awesome, thanks for testing!
What does `maverick info` show?

cglusky, 2017-11-19:
Maverick Environment:    dev 

Hardware platform not recognised
 ----------------------------------
OS:                     ubilinux
OS Version:             9.0
Architecture:           amd64
Kernel:                 4.9.45-ubilinux+
Timezone:               GMT
FQDN:                   maverick-up.home
IP Address:             192.168.2.67

now it says unrecognized hardware.  i did run a self-update

fnoop, 2017-11-19:
Wierd about arducopter using 95% cpu, it's using ~23% on up core.  If maverick-vision is using 50% cpu it must be transcoding in software.  I'll work on the hardware encoder

fnoop, 2017-11-19:
What does running this output?
```~/software/maverick/manifests/maverick-modules/maverick_hardware/facts.d/up.py```

fnoop, 2017-11-19:
Oh wait, does maverick say 1.1.4?
The imaging process resets it to stable branch. Change ~/data/config/maverick/maverick-branch.conf from stable to master and run self-update again

cglusky, 2017-11-19:
got it.  standby....

cglusky, 2017-11-19:
running configure after i reverted to master broke realsense...

[mNotice: Scope(Oncevcsrepo[git-realsense-realsense_sdk]): oncevcsrepo: /srv/maverick/var/build/realsense-sdk git repo doesn't exist locally, cloning may take a while..[0m
[1;33mWarning: Scope(Maverick_network::Process_interface[wman0]): WARNING: Interface config has changed significantly, PLEASE REBOOT TO ACTIVATE[0m
[mNotice: Compiled catalog for maverick-up.home in environment dev in 15.73 seconds[0m
[mNotice: /Stage[bootstrap]/Hosts/Host[maverick-up.home]/ip: ip changed '192.168.1.240' to '192.168.2.67'[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/File[/etc/profile.d/70-maverick-librealsense-cmake.sh]/ensure: defined content as '{md5}1cfae81a03eec12f7fe3bdbad3ff3b7b'[0m
[mNotice: /Stage[main]/Maverick_network/File[/etc/wpa_supplicant/wpa_supplicant.conf]/mode: mode changed '0644' to '0600'[0m
[mNotice: /Stage[main]/Maverick_hardware::Peripheral::Realsense/Package[freeglut3-dev]/ensure: created[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/Oncevcsrepo[git-realsense-realsense_sdk]/File[/srv/maverick/var/build/realsense-sdk]/ensure: created[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/Oncevcsrepo[git-realsense-realsense_sdk]/Vcsrepo[/srv/maverick/var/build/realsense-sdk]/ensure: created[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/File[/srv/maverick/var/build/realsense-sdk/build]/ensure: created[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/Exec[realsense-sdk-prepbuild]/returns: executed successfully[0m
[1;31mError: '/usr/bin/make -j4 >/srv/maverick/var/log/build/realsense-sdk.build.out 2>&1' returned 2 instead of one of [0][0m
[1;31mError: /Stage[main]/Maverick_vision::Realsense/Exec[realsense-sdk-build]/returns: change from 'notrun' to ['0'] failed: '/usr/bin/make -j4 >/srv/maverick/var/log/build/realsense-sdk.build.out 2>&1' returned 2 instead of one of [0][0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/Exec[realsense-sdk-install]: Dependency Exec[realsense-sdk-build] has failures: true[0m
[1;33mWarning: /Stage[main]/Maverick_vision::Realsense/Exec[realsense-sdk-install]: Skipping because of failed dependencies[0m
[mNotice: /Stage[main]/Maverick_vision::Realsense/File[/srv/maverick/var/build/.install_flag_realsense_sdk]: Dependency Exec[realsense-sdk-build] has failures: true[0m
[1;33mWarning: /Stage[main]/Maverick_vision::Realsense/File[/srv/maverick/var/build/.install_flag_realsense_sdk]: Skipping because of failed dependencies[0m
[mNotice: /Stage[main]/Maverick_network/Concat[/etc/udev/rules.d/10-network-customnames.rules]/File[/etc/udev/rules.d/10-network-customnames.rules]/content: content changed '{md5}8974b6dbeee4afe8b6280e33227643f4' to '{md5}844244c80c99b41c838b0b9043b9b5e6'[0m
[mNotice: /Stage[main]/Maverick_network/Network::Interface[lo]/Concat[/etc/network/interfaces]/File[/etc/network/interfaces]/content: content changed '{md5}685b5be8dde2a248cd7ff6ac54305ce5' to '{md5}93b841cf0056c5b6fd58a718ed2834be'[0m
[mNotice: /Stage[main]/Maverick_fc/Maverick_ros::Rosmaster[fc]/Service_wrapper[maverick-rosmaster@fc]/Service[maverick-rosmaster@fc]/ensure: ensure changed 'stopped' to 'running'[0m
[mNotice: /Stage[main]/Maverick_fc/Maverick_ros::Mavros[fc]/Service_wrapper[maverick-mavros@fc]/Service[maverick-mavros@fc]/ensure: ensure changed 'stopped' to 'running'[0m
[mNotice: /Stage[main]/Maverick_dev::Sitl/Maverick_ros::Rosmaster[sitl]/Service_wrapper[maverick-rosmaster@sitl]/Service[maverick-rosmaster@sitl]/ensure: ensure changed 'stopped' to 'running'[0m
[mNotice: /Stage[main]/Maverick_dev::Sitl/Maverick_ros::Mavros[sitl]/Service_wrapper[maverick-mavros@sitl]/Service[maverick-mavros@sitl]/ensure: ensure changed 'stopped' to 'running'[0m
[mNotice: Applied catalog in 54.99 seconds[0m

fnoop, 2017-11-19:
Ah it shouldn't be trying to install the realsense sdk - I think that's essentially dead now.  I'll remove it.  librealsense is the main library, but there are two versions now that need to be addressed.  The original deals with the older devices like the R200, then there's a new librealsense2 for the newer range.

cglusky, 2017-11-19:
ok.  can't get r200 to stream now.  but elp will stream YUYV at 9fps 1280x720 and takes right around 40% cpu

fnoop, 2017-11-20:
Add clean ubilinux user account in preprelease

fnoop, 2017-11-20:
Turned off deprecated realsense sdk in https://github.com/fnoop/maverick/commit/64c04161e2a203530eaf86dcb7bf466e4ce87680

fnoop, 2017-11-20:
Temperature missing from graphing, power consumption graphing ok now

fnoop, 2017-11-20:
/sys/class/thermal/thermal_zone0/temp is zero, not providing temps

fnoop, 2017-11-20:
https://up-community.org/forum/public-ubilinux/2467-no-thermal-values

cglusky, 2017-11-20:
@fnoop when you have a new iso ready let me know and i'll try a fresh install.

fnoop, 2017-11-21:
@cglusky Could you try a self-update and configure?  I probably won't create another iso until the 1.1.5 release

fnoop, 2017-11-21:
ubilinux keeps going to sleep, need to turn off auto suspend.

fnoop, 2017-11-21:
Also doesn't reboot cleanly

cglusky, 2017-11-22:
will do.  i did notice that the up^2 was taking some time to reboot.

cglusky, 2017-11-23:
configure yields this error...

```
[mNotice: Compiled catalog for maverick-up.home in environment dev in 15.62 seconds[0m
[1;31mError: Found 1 dependency cycle:
(Anchor[apt_key 418A7F2FB0E1E6E7EABF6FE8C2E73424D59097AB present] => Apt::Key[Add key: 418A7F2FB0E1E6E7EABF6FE8C2E73424D59097AB from Apt::Source grafana] => Apt::Setting[list-grafana] => File[/etc/apt/sources.list.d/grafana.list] => Class[Apt::Update] => Exec[apt_update] => Class[Apt::Update] => Stage[bootstrap] => Stage[main] => Class[Maverick_ros] => Exec[ros-repo] => Exec[apt_update])
Try the '--graph' option and opening the resulting '.dot' file in OmniGraffle or GraphViz[0m
[1;31mError: Failed to apply catalog: One or more resource dependency cycles detected in graph[0m
```

Reboot takes perhaps 20 seconds but it does reboot

fnoop, 2017-11-24:
Yeah OK the reboot is a little slow but it's not too bad.  I suspect it's just services shutting down like influxdb, which can be very slow to flush everything and shut down.
The configure error is because the master branch is under heavy development at the moment, refactoring for a new environment.  Will fix it today.

fnoop, 2017-11-24:
Thermal values solved here:
https://up-community.org/forum/public-ubilinux/2467-no-thermal-values

closed 2017-11-24 by fnoop