goodrobots/maverick #696: Gstreamer vaapi not installed on intel platforms

fnoop, 2017-11-29:
From #640, minnowboard/up boards etc - basically all intel boards should install vaapi support.

fnoop, 2017-11-30:
Refactored manifests to support intel platforms generically, added OpenCL gpu (beignet) and VAAPI support.  01org/gfx updated driver support only for Ubuntu.
@cglusky this should enable support for minnowboard and for up squared.

closed 2017-11-30 by fnoop

cglusky, 2017-11-30:
fantastic!  will fire them up in the next few days to test and report back.

cglusky, 2017-12-03:
NO luck with vaapi on Up^2.  Assuming it's not related to Beignet compile fail.  Or it could be me.  I am testing by adding this to conf 

```
# Encoder, if not specified will default to h264.  Values are h264, mjpeg or none
encoder = vaapih264enc
```

Yields this in log...
`2017-12-02 23:58:43,755 - CRITICAL - Stream starting with unrecognised encoder: vaapih264enc`

Will have MinnowBoard back up in a few days and try on Ubuntu.

fnoop, 2017-12-03:
vaapi is not an encoder type in visiond - that's h264.  If vaapi is present in gstreamer it will (should) automatically find it and use it (and you'll see a note in the log).  Change the encode back to h264, or just leave it blank.

fnoop, 2017-12-03:
Test if vaapis is installed with:
```gst-inspect-1.0 |grep vaapi```