goodrobots/maverick #701: Beignet Compile Fails

cglusky, 2017-12-02:
Up^2 self-update to master 1.1.5-dev

Log attached.

[maverick.dev.2017-12-02-20-49.log](https://github.com/fnoop/maverick/files/1524162/maverick.dev.2017-12-02-20-49.log)

fnoop, 2017-12-02:
Could you please attach /srv/maverick/var/log/build/beignet.cmake.out?

cglusky, 2017-12-02:
attached...

[beignet.cmake.out.txt](https://github.com/fnoop/maverick/files/1524340/beignet.cmake.out.txt)

[CMakeError.log](https://github.com/fnoop/maverick/files/1524343/CMakeError.log)

[CMakeOutput.log](https://github.com/fnoop/maverick/files/1524344/CMakeOutput.log)

fnoop, 2017-12-03:
That's odd, I thought I had fixed that.  Can you see if you have the dependencies installed?
```
dpkg -l |grep clang
dpkg -l |grep llvm
```

fnoop, 2017-12-03:
nvm I've replicated.

fnoop, 2017-12-03:
Should be fixed now

closed 2017-12-03 by fnoop