goodrobots/maverick #702: Implement ROS2

fnoop, 2017-12-03:


fnoop, 2017-12-03:
https://github.com/ros2/ros2/wiki/Linux-Development-Setup

fnoop, 2019-09-11:
Update to latest ros2, build with our custom python3, and build/install by default.

SamuelDudley, 2019-09-11:
Do you expect this to impact macros?

SamuelDudley, 2019-09-11:
It looks like ROS2 development has not occurred:
https://github.com/mavlink/mavros/issues/771

fnoop, 2019-09-16:
When building with custom python3.7.4, some python modules can't be installed as dependencies:
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp
Traceback (most recent call last):
  File "/srv/maverick/var/build/ros2/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 50, in get_sip_dir_flags
    sip_dir = config.pyqt_sip_dir
  File "/srv/maverick/software/python/lib/python3.7/site-packages/sipconfig.py", line 192, in __getattr__
    raise AttributeError("\"%s\" is not a valid configuration value or user option" % name)
AttributeError: "pyqt_sip_dir" is not a valid configuration value or user option

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/maverick/var/build/ros2/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 85, in <module>
    sip_dir, sip_flags = get_sip_dir_flags(config)
  File "/srv/maverick/var/build/ros2/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 65, in get_sip_dir_flags
    raise FileNotFoundError('The sip directory for PyQt5 could not be located. Please ensure' +
FileNotFoundError: The sip directory for PyQt5 could not be located. Please ensure that PyQt5 is installed
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp	[ Exited with code 2 ]
[Processing: qt_gui_cpp, rviz_ogre_vendor]
Aborted  <<< rviz_ogre_vendor
```

SamuelDudley, 2019-09-16:
so pyqt is failing which is a GUI lib. As it is likely that people wont be running GUI applications on maverick is it safe to not build these components?
Is there a headless flag or similar that avoids building the GUI modules?

fnoop, 2019-09-16:
I started trying to remove all the qt modules but gave up after a while because there were so many of them.  There doesn't appear to be anything in the docs for building a non-desktop variant.

Building the UI stuff for ROS is very useful as it includes rviz and other tools that ros devs commonly use, assuming rviz is still a thing in ros2.

SamuelDudley, 2019-09-16:
Okay, understood. looking at ROS1 installs there was an option to install the base or core packages and another option to install the desktop build. As you said such an option with the ROS2 install didn't jump out at me...

fnoop, 2019-09-17:
Looks like we can use ros variants with ros2 to extract ros-base packages, which excludes the desktop modules.
https://github.com/ros2/variants

fnoop, 2019-09-17:
ros_base builds successfully with our custom python :)
need to figure out how to build ros1_bridge.

SamuelDudley, 2019-09-17:
Nice!

SamuelDudley, 2019-09-17:
They don't make it easy to work out what packages you need to build it with ros_base...

fnoop, 2019-09-17:
You can use rosinstall_generator like this:
`/srv/maverick/software/python/bin/rosinstall_generator ros_base --rosdistro dashing --deps >ros2.repos`

fnoop, 2019-09-17:
Hmm, even though it seems to be using maverick python, it's creating ~/software/ros2/current/lib/python3.6 packages :(

For ros1_bridge environment:
https://ask.puppet.com/question/2177/how-to-execute-source-command/

fnoop, 2019-09-17:
Fixed build process to use only custom python3.7.

fnoop, 2019-09-17:
Working on ros1_bridge.  Needs partial ros1 and ros2 environment set to build which is tricky in puppet that deliberately blanks all environment.
https://github.com/ros2/ros1_bridge/blob/master/README.md#building-the-bridge-from-source

fnoop, 2019-09-17:
ros1_bridge quite limited in usefulness:
https://github.com/ros2/ros1_bridge#bridge-communication-between-ros-1-and-ros-2

support for mavros, realsense etc still much better in ros1, and -api written for ros1, so will leave ros2 until better supported and required.

for now, it's installed by default at least.

closed 2019-09-17 by fnoop