goodrobots/maverick #706: Support Nvidia Tegra platform

fnoop, 2017-12-07:


fnoop, 2017-12-17:
Visionworks = OpenVX.  Good.
opencv4tegra = ancient 2.4 + cuda stuff.  avoid
cudnn/CUDA stuff = proprietary cuda.  avoid except where necessary to support opencl/openvx/opencv layers.

fnoop, 2017-12-17:
TX1 only has 16gb mmc, less than half left after installing jetpack crap

fnoop, 2017-12-17:
Lots of odd iptables errors:
```
Error: /Stage[main]/Maverick_web/Maverick_security::Firewall::Firerule[webserver]/Firewall[100 allow webserver access from []]/ensure: change from 'absent' to 'present' failed: Execution of '/sbin/iptables -I INPUT 4 --wait -t filter -p tcp -m multiport --dports 80,443 -m comment --comment 100 allow webserver access from [] -j ACCEPT' returned 1: iptables: No chain/target/match by that name.
```

fnoop, 2017-12-17:
^^ Kernel is missing multiport iptables support..

fnoop, 2017-12-17:
Collectd: Add power consumption and cuda data

fnoop, 2017-12-17:
Opencv: Add opencl/openvx support

fnoop, 2017-12-17:
Gstreamer: add omx and nvcamerasrc support

fnoop, 2017-12-17:
Probably nvcamerasrc support isn't needed any more because device is exposed in v4l2

fnoop, 2017-12-17:
No opencl for tegra :(
https://devtalk.nvidia.com/default/topic/974021/jetson-tx1/can-i-use-opencl-in-jetson-tx1-/

fnoop, 2017-12-17:
Power consumption:
https://stackoverflow.com/questions/44228005/how-to-measure-power-consumption-of-jetson-tx1

fnoop, 2017-12-17:
Power consumption on TX1 module only supported by later >300 revision :(  Development board power monitors available through i2c but that's no good once the module is on the carrier board.  Does j120 have power monitor available through i2c?
Here's a possible alternative:
https://github.com/cezs/jtx1inst

fnoop, 2017-12-17:
jtx1inst pulls lots of interesitng power and temp data

fnoop, 2017-12-18:
nvidia-smi not available on tegra so collectd_cuda doesn't work.  Will have to write a plugin, stats available:
https://devtalk.nvidia.com/default/topic/928386/gpu-usage-monitoring-on-tx1/

fnoop, 2017-12-18:
How to build tensorflow:
http://www.jetsonhacks.com/2017/09/18/build-tensorflow-on-nvidia-jetson-tx1-development-kit/

fnoop, 2017-12-18:
Build flash image:
https://devtalk.nvidia.com/default/topic/1027208/jetson-tx1/problems-installing-jetpack-onto-sd-card-tx1-/
http://docs.nvidia.com/jetpack-l4t/3.1/index.html#developertools/mobile/jetpack/l4t/3.1/jetpack_l4t_install.htm
https://elinux.org/Boot_from_sd

fnoop, 2017-12-19:
Basic tegra support added, and tx1/tx2 download images published.  Bumping this to 2.0 for future development.

SamuelDudley, 2017-12-21:
Some notes from gitter: 
- extract the image as root to allow for correct permissions in the flashed image
- hardware serial address is incorrect causing fc modules that require the mavlink router to crash
- flash process works fine on the j120 carrier board

fnoop, 2017-12-21:
Add openkai support

fnoop, 2017-12-21:
jtx1inst reports module power when running on j120 carrier:
```
[dev] [mav@maverick-tegra ~/var/lib/jtx1inst/build]$ sudo ./jtx1inst_demo
[POWER] module power input: 2551.000mW
[POWER] GPU power rail: 1039.000mW
[POWER] CPU power rail: 1039.000mW
Error opening file: No such file or directory
```
Much less info than on the dev board, but still very useful.

No i2c power monitor devices show up in the kernel sysfs nodes. jtx1inst has a different method to directly access the INA.

fnoop, 2017-12-21:
Detailed clock frequency available through jetson utility:
```
[dev] [mav@maverick-tegra /home/nvidia]$ sudo ./jetson_clocks.sh --show
SOC family:tegra210  Machine:jetson_tx1
Online CPUs: 0-3
CPU Cluster Switching: Disabled
cpu0: Gonvernor=interactive MinFreq=102000 MaxFreq=1734000 CurrentFreq=1224000
cpu1: Gonvernor=interactive MinFreq=102000 MaxFreq=1734000 CurrentFreq=1020000
cpu2: Gonvernor=interactive MinFreq=102000 MaxFreq=1734000 CurrentFreq=1020000
cpu3: Gonvernor=interactive MinFreq=102000 MaxFreq=1734000 CurrentFreq=1224000
GPU MinFreq=76800000 MaxFreq=998400000 CurrentFreq=76800000
EMC MinFreq=12750000 MaxFreq=1600000000 CurrentFreq=1065600000 FreqOverride=0
Can't access Fan!
```

fnoop, 2017-12-21:
More stats including CUDA usage available from tegrastats:
```
[dev] [mav@maverick-tegra /home/nvidia]$ sudo ./tegrastats
RAM 735/3983MB (lfb 669x4MB) cpu [0%,0%,0%,0%]@816 EMC 0%@665 APE 25 GR3D 0%@76
RAM 735/3983MB (lfb 669x4MB) cpu [12%,24%,3%,4%]@306 EMC 3%@102 APE 25 GR3D 0%@76
RAM 735/3983MB (lfb 669x4MB) cpu [4%,34%,3%,6%]@204 EMC 4%@68 APE 25 GR3D 0%@76
RAM 736/3983MB (lfb 669x4MB) cpu [9%,37%,6%,5%]@204 EMC 4%@68 APE 25 GR3D 0%@76
RAM 736/3983MB (lfb 669x4MB) cpu [6%,37%,6%,6%]@204 EMC 3%@68 APE 25 GR3D 0%@76
RAM 736/3983MB (lfb 669x4MB) cpu [7%,36%,3%,6%]@306 EMC 2%@102 APE 25 GR3D 0%@76
```

fnoop, 2017-12-21:
Multiple UARTs exposed.  Console started automatically on ttyS0:
```
[    0.540016] console [ttyS0] disabled
[    0.540090] 70006000.serial: ttyS0 at MMIO 0x70006000 (irq = 320, base_baud = 25500000) is a Tegra
[    2.556337] console [ttyS0] enabled
[    2.557711] 70006040.serial: ttyTHS1 at MMIO 0x70006040 (irq = 321, base_baud = 0) is a TEGRA_UART
[    2.558180] 70006200.serial: ttyTHS2 at MMIO 0x70006200 (irq = 322, base_baud = 0) is a TEGRA_UART
[    2.558650] 70006300.serial: ttyTHS3 at MMIO 0x70006300 (irq = 323, base_baud = 0) is a TEGRA_UART
```

fnoop, 2017-12-21:
Tensorflow binaries:
https://github.com/jetsonhacks/installTensorFlowJetsonTX

fnoop, 2017-12-23:
`maverick configure` hangs/resets tx1 every time if `"maverick_fc::mavlink_input": "/dev/ttyTHS1"` is set.  If set back to default (/dev/ttyAMA), no hang.  Suspect this must be a kernel bug.

fnoop, 2017-12-23:
Ah:
https://github.com/NVIDIA-Jetson/redtail/tree/master/tools/install/tx1-uart-patch

fnoop, 2017-12-23:
This is cool, should add:
https://github.com/NVIDIA-Jetson/redtail

fnoop, 2017-12-23:
TX2 patch:
https://github.com/NVIDIA-Jetson/redtail/tree/master/tools/install/j120-tx2-patch

fnoop, 2017-12-23:
Maverick tegra image takes a vast amount of disk space to flash as it constructs the system.img.raw and then the system.img before flashing the system.img - needs about 40Gb free on the flashing system.  Can pre-create the system.img with a blank rootfs so all that disk space isn't necessary, also much faster to unpack the single file than the rootfs and faster to flash as no need to recreate the image.

closed 2018-06-01 by fnoop