goodrobots/maverick #711: Add mapping software

fnoop, 2017-12-29:
https://webodm.org/
https://github.com/OpenDroneMap/WebODM
https://github.com/OpenDroneMap/OpenDroneMap/

SamuelDudley, 2018-01-01:
Interesting blog which has related content: http://gallinazo.flightgear.org/uas/generating-survey-area-coverage-routes/
http://gallinazo.flightgear.org/software/image-analysis/image-stitching-tutorial-part-1-introduction/

I would LOVE to see some online stitching action to workout coverage maps and automatic mission replanning. That's my dream anyway!

cglusky, 2018-01-03:
@SamuelDudley I see you listed as a contributor for image analysis.  i am a little confused by what they are doing and what is being done in ODM - is there a big difference?

SamuelDudley, 2018-01-03:
I agree that supporting ODM / web ODM is the way to go as the image analysis stuff is all a bit academic at this point. I think there are some useful tools in the lib though which would help with data preparation prior for use in ODM. Frame extraction from movies, alignment of autopilot logs and camera frames, automated mission planning, effecent representation of mission information,  etc... mainly I have found the blog to be a good read if your interested in that sort of stuff.

My contribution was only a little bug fix in the included SRTM lib, nothing ground breaking :)

cglusky, 2018-01-07:
that is a good read.  the aura work he is doing is also interesting.

fnoop, 2018-01-14:
https://github.com/hotosm/OpenAerialMap

cglusky, 2018-01-14:
The communities around grass roots mapping have been around awhile.  I follow them loosely these days.  There is a project that may have some crossover to what we are trying to do:  https://github.com/posm/posm

cglusky, 2018-01-14:
Note the fairly heavy hardware requirements.  Just the reality of doing this type of work in the field.  And if you check out the POSM org on github they have an admin UI that may be interesting to study.  Have not looked at it in detail.