goodrobots/maverick #716: ZeroTier not installed by default on RPI3

Al3x82, 2018-01-05:
Missing the config string inside the documentation  "maverick_network::zerotier::libzt" = true,

fnoop, 2018-01-05:
libzt support isn't working/complete yet, so hasn't been documented.

fnoop, 2018-01-05:
Zerotier should be installed though, are you saying it's not on the raspberry?  As the docs say you need to set active parameter in order to turn it on:
`"maverick_network::zerotier::active": true`

Al3x82, 2018-01-06:
> Zerotier should be installed though, are you saying it's not on the raspberry?

Yes I think so, these were my steps:

- Installed OS version with etcher

- Selfupdated and configured

- Added to config `"maverick_network::zerotier::active": true`  `"maverick_network::zerotier::network_id": "myprivatenetworkid"` 

- configured
At this point i got some compiling errors, chcked in my private network page, and nothing connected.

- added to config ` "maverick_network::zerotier::libzt" = true,`

- configured
Downloaded and compiled without errors, and now is connected to my private network.

ifconfig log : 

`zt0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 2800
        inet 29.39.76.63  netmask 254.0.0.0  broadcast 29.255.255.255
        inet6 fd80:56c2:e21c:0:199:931f:fb27:4ac7  prefixlen 88  scopeid 0x0<glo
        inet6 fe80::1f:fbff:fe3b:a805  prefixlen 64  scopeid 0x20<link>
        ether 02:1f:fb:3b:a8:05  txqueuelen 1000  (Ethernet)
        RX packets 1486  bytes 143888 (140.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11  bytes 966 (966.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

zt1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 2800
        inet 10.244.60.70  netmask 255.255.0.0  broadcast 10.244.255.255
        inet6 fc63:f262:3d1f:fb27:4ac7::1  prefixlen 40  scopeid 0x0<global>
        inet6 fe80::2037:a5ff:fe56:548d  prefixlen 64  scopeid 0x20<link>
        ether 22:37:a5:56:54:8d  txqueuelen 1000  (Ethernet)
        RX packets 6885  bytes 1747876 (1.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7220  bytes 10581512 (10.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`

fnoop, 2018-01-06:
I've just done a fresh download and install of the raspberry (not raspberrylite) image, self-update and configure, added `"maverick_network::zerotier::active": true` and it works fine.  Zerotier is already installed so it shouldn't be a compile error you're getting, probably a configure error.  Adding the libzt entry won't have made any difference, as it's not currently working or used by anything.

Could you possibly find the maverick log entry in ~/var/log/maverick where you saw the error, and post here?

Al3x82, 2018-01-07:
I found only installation entry in my log, these are few lines:

^[[mNotice: Scope(Oncevcsrepo[git-libzt]): oncevcsrepo: /srv/maverick/software/libzt git repo doesn't exist locally, cloning may take a while..^[[0m
^[[mNotice: Compiled catalog for maverick-raspberry.home in environment dev in 23.65 seconds^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Package[swig]/ensure: created^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Package[swig-examples]/ensure: created^[[0m
^[[mNotice: /Stage[main]/Maverick_vision::Visiond/Service_wrapper[maverick-visiond]/Service[maverick-visiond]/ensure: ensure changed 'stopped' to 'running'^[[0m
^[[mNotice: /Stage[main]/Grafana::Install/Archive[/tmp/grafana.deb]/ensure: download archive from https://dl.bintray.com/fg2it/deb-rpi-1b/main/g/grafana_4.6.1_armhf.deb to /tmp/grafana.deb  with cleanup^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Oncevcsrepo[git-libzt]/File[/srv/maverick/software/libzt]/ensure: created^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Oncevcsrepo[git-libzt]/Vcsrepo[/srv/maverick/software/libzt]/ensure: created^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: echo LIBZT_IPV6=1 IPV6=1 LIBZT_IPV4=1 IPV4=1^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: g++ -Ofast -fstack-protector -Wno-format -fno-rtti -std=c++11 -c src/lwIP.cpp \^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:   -DZT_SDK -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -Iext -Izto/osdep -Izto/node -Izto/service -Izto/include  -Iext/lwip-contrib/ports/unix/include -Iext/lwip/src/include/lwip -Iext/lwip/src/include -I/include -Iext/lwip/src/include/ipv$
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: LIBZT_IPV6=1 IPV6=1 LIBZT_IPV4=1 IPV4=1^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: make -f make-liblwip.mk liblwip.a LIBZT_IPV6=1 IPV6=1 LIBZT_IPV4=1 IPV4=1^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: make[1]: Entering directory '/srv/maverick/software/libzt'^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: g++ -Ofast -fstack-protector -Wno-format -fno-rtti -std=c++11 -c src/VirtualSocket.cpp \^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:   -Iinclude -I../include -Isrc^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: make-liblwip.mk:101: .depend: No such file or directory^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: g++  -Wno-format -Wno-deprecated -O3 -g -Wall -fPIC -Iext/lwip/src/include -Iext/lwip-contrib/ports/unix -Iext/lwip-contrib/ports/unix/include -Iext/lwip/src -I. -Iext -Iinclude -DLWIP_IPV4=1 -DIPv4 -DLWIP_IPV6=1 -DIPv6 -MM ext/lwip/src/c$
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: g++ -Ofast -fstack-protector -Wno-format -fno-rtti -std=c++11 -c src/VirtualBindingPair.cpp \^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:   -Iext -Izto/osdep -Izto/node -Izto/service -Izto/include  -Iinclude -I../include -Isrc^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: g++ -Ofast -fstack-protector -Wno-format -fno-rtti -std=c++11 -c src/VirtualTap.cpp \^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:   -DZT_SDK -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -Iext -Izto/osdep -Izto/node -Izto/service -Izto/include   -Iinclude -I../include -Isrc -DLWIP_DONT_PROVIDE_BYTEORDER_FUNCTIONS -DSTACK_LWIP -DLIBZT_IPV4 -DLWIP_IPV4=1 -DLIBZT_IPV6 -DL$
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: In file included from include/libzt.h:45:0,^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:                  from src/VirtualTap.cpp:43:^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: include/Defs.h:108:0: warning: "IP_BIND_ADDRESS_NO_PORT" redefined^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  #define IP_BIND_ADDRESS_NO_PORT 201^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  ^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: In file included from /usr/include/netinet/in.h:37:0,^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:                  from /usr/include/arpa/inet.h:22,^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:                  from zto/osdep/Phy.hpp:62,^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:                  from src/VirtualTap.cpp:35:^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: /usr/include/arm-linux-gnueabihf/bits/in.h:103:0: note: this is the location of the previous definition^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  #define IP_BIND_ADDRESS_NO_PORT 24^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  ^[[0m
[... cut ...]
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:                  from src/VirtualTap.cpp:35:^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns: /usr/include/arm-linux-gnueabihf/bits/in.h:101:0: note: this is the location of the previous definition^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  #define IP_NODEFRAG     22^[[0m
^[[mNotice: /Stage[main]/Maverick_network::Zerotier/Exec[libzt-make]/returns:  ^[[0m

....continue

fnoop, 2018-01-07:
I meant the run before that, where the zerotier fails.  libzt is not relevant, it's not working or used currently.

Al3x82, 2018-01-07:
I cannot find any log related to zerotier before. My apologize if this issue have bothered you!
Anyway you are doing a great job with maverick!

fnoop, 2018-01-08:
No problem, thanks for the feedback!  If you ever see this again, please do try and capture the log from ~/var/log/maverick that relates to the failure and post here.  I suspect what you saw was not a compile problem or a problem with Maverick, but probably some failure to connect to zerotier.
Glad you got it working in the end!

closed 2018-01-08 by fnoop