goodrobots/maverick #720: Tegra out of range

rbachtell, 2018-01-20:
[maverick.dev.2018-01-20-21-22.log](https://github.com/goodrobots/maverick/files/1649321/maverick.dev.2018-01-20-21-22.log)

rbachtell, 2018-01-20:
[maverick.dev.2018-01-20-22-20.log](https://github.com/goodrobots/maverick/files/1649338/maverick.dev.2018-01-20-22-20.log)

fnoop, 2018-01-22:
@rbachtell apologies I missed your verbose log upload.  The last entry is trying to turn off the desktop.  Could you try adding the following to your localconf.json and see if it makes any difference?
"maverick_desktop::enable": true
If you could do a normal run, and then if it still hangs another verbose run and upload the log, would be very helpful.  Thanks.

rbachtell, 2018-01-23:
That has cured the hang but I now get a long error message just before the “happy flying”  I can’t figure out how to copy it to the SD card but it starts out: Error:no parameter named ‘ensures’ at /srv/maverick/software/maverick/manifests/maverick-modules/maverick_desktop/manifests/init.pp:47

rbachtell, 2018-01-23:
It hangs when I shut down😒

fnoop, 2018-06-01:
I think this should be fixed by previous maverick_desktop updates, please reopen if still an issue

closed 2018-06-01 by fnoop